tpaviot/pythonocc-core #610: Wrapper of ShapeAnalysis_Curve.Project()

daha96, 2019-01-09:
Hi,

I'm stuck on the projection from a point on a BRepAdaptor_CompCurve. I tried a lot to make the ShapeAnalysis_Curve().Project()-method run, but I get following error:

    distance = ShapeAnalysis_Curve().Project(curve, pnt, tolerance, proj, parameter, True)
    TypeError: in method 'ShapeAnalysis_Curve_Project', argument 2 of type 'Handle_Geom_Curve const &'

Here is the code:

    pnt = gp_Pnt(1, 2, 3)
    curve = BRepAdaptor_CompCurve(wire)
    parameter = 0.
    tolerance = 1e-7
    proj = gp_Pnt()
    distance = ShapeAnalysis_Curve().Project(curve, pnt, tolerance, proj, parameter, True)

I can't find wrong parameters while degugging, is it possible, there is somthing wrong in the wrapper?

Kind regards,
Daniel

tpaviot, 2019-01-09:
There's something wrong with the type of the ```curve``` parameter you pass. Can you please post a full running example so that I can test on my machine.

tpaviot, 2019-01-09:
You can also try to pass ```curve = BRepAdaptor_CompCurve(wire).Curve()``` or even ```curve = BRepAdaptor_CompCurve(wire).Curve().Curve()```

daha96, 2019-01-09:
`curve = BRepAdaptor_CompCurve(wire).Curve()` doesn't work either. I get this error:

    AttributeError: 'BRepAdaptor_CompCurve' object has no attribute 'Curve'

The sample above is almost the full running example. All I do before is loading the wire from a file:

    from OCC.BRep import BRep_Builder
    from OCC.BRepTools import breptools
    from OCC.BRepAdaptor import BRepAdaptor_CompCurve
    from OCC.ShapeAnalysis import ShapeAnalysis_Curve
    from OCC.TopoDS import TopoDS_Shape, topods
    from OCC.gp import gp_Pnt
    
    # Read wire
    wire = TopoDS_Shape()
    aBuilder = BRep_Builder()
    breptools.Read(wire, "wire.txt", aBuilder)
    wire = topods.Wire(wire)
    
    # Project point on curve
    pnt = gp_Pnt(1, 2, 3)
    curve = BRepAdaptor_CompCurve(wire)
    parameter = 0.
    tolerance = 1e-7
    proj = gp_Pnt()
    distance = ShapeAnalysis_Curve().Project(curve, pnt, tolerance, proj, parameter, True)

[wire.txt](https://github.com/tpaviot/pythonocc-core/files/2741550/wire.txt)

tpaviot, 2019-01-09:
Have a look at the core_geometry_project_point_on_curve.py demo here https://github.com/tpaviot/pythonocc-demos/blob/master/examples/core_geometry_project_point_on_curve.py

daha96, 2019-01-09:
I read this demo, but I think, 'GeomAPI_ProjectPointOnCurve' works only with edges, not with wires. Thats the reason why I choose 'ShapeAnalysis_Curve().Project()'

tpaviot, 2019-01-10:
The issue you report is related to a misuse of the Project method, not to an issue with the wrapper. You can have a look at https://www.opencascade.com/content/topodswire-topodsedge . Here you can find how to convert a TopoDS_Wire to a Geom_Curve (required by the Project method of ShapeAnalaysis_Curve).

I fixed your code and added a related example at https://github.com/tpaviot/pythonocc-demos/commit/c2c062ae7aa0e1bbdd8f7050dbe728c6c31eb97e

closed 2019-01-10 by tpaviot

daha96, 2019-01-13:
Thanks, it works!