tpaviot/pythonocc-core #612: Twisted fillet on fused shape

pyoccMAN, 2019-01-11:
Hello,

I am trying to make a fillet on fused shape.  I have used the SetRadius option but the fillet is twisted:
[image]
 I am not sure how SetRadius works. One first specifies radius, and than the contour index, but the third parameter I am not sure.

Thank you.

tpaviot, 2019-01-11:
Hard to say what's going on without more details.

pyoccMAN, 2019-01-11:
Alos, some surfaces are missing:
[image]

pyoccMAN, 2019-01-11:
The contour is shown in the picture below:
[image]

tpaviot, 2019-01-11:
```SetRadius``` is a class method. Which class to you use to perform the fille ? BRepFilletAPI_MakeFillet ?

You can have a look at fillet related examples here https://github.com/tpaviot/pythonocc-demos/blob/master/examples/core_topology_fillet.py

pyoccMAN, 2019-01-11:
Yes, I use BRepFilletAPI_MakeFillet. I had a look at both examples regarding fillets. Add class method fails, and  SetRadius makes the fillets but twisted. I am not sure what is :type IinC: int reffered in the SetRadius. Is there another suitable class for fillets? Can I specify tangent for fillets?

tpaviot, 2019-01-12:
Documentation related to chamfers and fillets can be read at https://www.opencascade.com/doc/occt-6.9.1/overview/html/occt_user_guides__modeling_algos.html#occt_modalg_6

tpaviot, 2019-01-12:
And the documentation related to the ```BRepFilletAPI_MakeFillet``` class https://www.opencascade.com/doc/occt-6.9.1/refman/html/class_b_rep_fillet_a_p_i___make_fillet.html

This is the best I can do, please report any progress.