tpaviot/pythonocc-core #619: PyQt/MacOS display issues

adam-urbanczyk, 2019-01-17:
I am not sure it is an PythonOCC issue, but run totally out of ideas. Maybe someone here had the same problem. It is described here: https://github.com/CadQuery/CQ-editor/issues/7. Executive summary: I am using PythonOCC display widget (PyQT5) in a GUI for the CadQuery project. It works on windows and linux. On mac the application starts but no 3d graphics is displayed:

[image]

Here are some relevant lines printed to stdout:

```
$ python run.py
INFO:OCC.Display.backend:backend loaded: qt-pyqt5
 ###### 3D rendering pipe initialisation #####
Display3d class initialization starting ...
Aspect_DisplayConnection created.
Graphic_Driver created.
V3d_Viewer created.
AIS_InteractiveContext created.
V3d_View created
Cocoa window created.
Display3d class successfully initialized.
 ########################################
Layer manager created
Layer dimensions: 100, 30
TKOpenGl | Type: Performance | ID: 0 | Severity: Low | Message:
  VBO creation for Primitive Array has failed for 4 vertices. Out of memory?
TKOpenGl | Type: Performance | ID: 0 | Severity: Low | Message:
  VBO creation for Primitive Array has failed for 169 vertices. Out of memory?
TKOpenGl | Type: Performance | ID: 0 | Severity: Low | Message:
  VBO creation for Primitive Array has failed for 26 vertices. Out of memory?
TKOpenGl | Type: Performance | ID: 0 | Severity: Low | Message:
  VBO creation for Primitive Array has failed for 26 vertices. Out of memory?
TKOpenGl | Type: Performance | ID: 0 | Severity: Low | Message:
  VBO creation for Primitive Array has failed for 26 vertices. Out of memory?
TKOpenGl | Type: Error | ID: 0 | Severity: High | Message:
  New texture intialization of size 16384x128 for textured font has failed.
TKOpenGl | Type: Error | ID: 0 | Severity: High | Message:
  Font 'Courier' - initialization of GL resources has failed!
TKOpenGl | Type: Performance | ID: 0 | Severity: Low | Message:
  VBO creation for Primitive Array has failed for 4 vertices. Out of memory?
```
I'll be grateful for any tips.

tpaviot, 2019-01-18:
Can you run the ```core_helloworld.py``` example on your OSX machine ? If you can, then this issue is not related to pythonocc but rather pyqt or the way you instanciate the viewer into your qt app.

On the other hand, can you please try to replace, in the ```qtDisplay.py``` module, the following line:
```
class qtBaseViewer(QtOpenGL.QGLWidget)
```
to
```
class qtBaseViewer(QtWidgets.QOpenGLWidget)
```
and report result ?

tpaviot, 2019-01-18:
This might help https://stackoverflow.com/questions/31520371/what-is-the-difference-between-qglwidget-and-qopenglwidget

adam-urbanczyk, 2019-01-19:
Thanks @tpaviot , we are investigating your suggestion. I don't have access to Mac so it is going rather slow.

adam-urbanczyk, 2019-01-20:
So here are the results (adding @jmwright for info - he did all the tests):
*  `core_helloworld.py`  does show a box;
* changing to QOpenGLWidget does not help.

adam-urbanczyk, 2019-04-26:
@bernhard-42 discovered that on Mac calling `InitDriver()` needs to happen after calling `show()`. This seems to be the solution. Is it expected/known behavior?

tpaviot, 2019-04-26:
Thankd @adam-urbanczyk for the feedback. I never got such issue reported before, do you mean that the SimpleGui bundled with pythonocc is also affected ? or only the cq-editor you build on top of pythonocc ?

tpaviot, 2019-04-26:
I just checked the qtDisplay.py module, it appears that, at pythonocc, we call the InitDriver method **after** the qt window is actually showed https://github.com/tpaviot/pythonocc-core/blob/master/src/Display/SimpleGui.py#L188
This is the case for all platforms.
So I guess it's a cq specific issue, isn't it ?

bernhard-42, 2019-04-26:
No worries, I learned from SimpleGui that show has to come before InitDriver
```
        win = MainWindow()
        win.show()
        win.resize(size[0], size[1])
        win.centerOnScreen()
        win.canva.InitDriver()
```

Since this worked on the Mac, I compared cq-gui's implementation and traced it down to the order show => InitDriver.

bernhard-42, 2019-04-26:
I know little to nothing about QT5, so no idea why it behaves differently on macos.

tpaviot, 2019-04-26:
What I learnt working with various GUI libraries (wx, qt, pysiqde) on different platforms (macos, linux, windows) is that you can't trust any "portability" a priori. Eveything has to be tested, a simple line move can make a big change, a small library upgrade can destroy weeks of work. When, by an improbable miracle, you get something working on all platforms as expected, don't touch it ever and consider it as an house of cards that will collapse as soon as you try to improve it.

bernhard-42, 2019-04-26:
Agreed, so I often try to use the browser (I mean Chrome, newest version ;-) ) as the "GUI framework".
My latest attempt is an occ cadquery viewer for Jupyter (https://github.com/bernhard-42/jupyter-cadquery) - where I btw. borrowed some of your threejs code - thank you - however, using it via pythreejs. But, I wasn't brave enough to test it on Firefox, Safari, ...

tpaviot, 2019-04-26:
Your jupyter viewer looks great. Maybe I'll try to borrow back some of your contributions to improve the pythonocc jupyter renderer.

bernhard-42, 2019-04-26:
You are welcome! If you have questions, let me know!

adam-urbanczyk, 2019-04-26:
@tpaviot indeed `SimpleGui` works fine. I was just wondering why is it necessary to call `InitDriver()` after `show()` (only on Mac). If one is making a larger application (like CQ-editor) with many components this is rather counterintuitive.

Anyhow issue is solved, so I'm closing it.

closed 2019-04-26 by adam-urbanczyk

bernhard-42, 2019-04-27:
@tpaviot I have refactored my viewer and added support for python-occ (without importing any cadquery libs). Some examples taken from your repo can be found in [OCC examples.ipynb]( https://github.com/bernhard-42/jupyter-cadquery/blob/master/examples/OCC%20examples.ipynb).
The screenshot of the classic OCC bottle can be found in [classic OCC bottle](https://github.com/bernhard-42/jupyter-cadquery/blob/master/screenshots/occ_bottle.png)