tpaviot/pythonocc-core #623: Problem exporting to vector formats

cristobaltapia, 2019-02-06:
I am having a strange problem with the vector-format export functionality. I have installed python-occ with conda under Ubuntu 18.10 and when I run the example to export the torus the respective file gets generated, but in the case of the pdf export, only the background is visible, and in the case of svg-format, a very bad results is obtained (see screenshots below). I have also manually compiled OCE, python-occ and gl2ps to be sure and still get the same results. Might this be hardware-related? If so, how?
I have a relatively good graphics card (AMD Radeon 550 RX) so that shouldn't be the problem, since it is good supported under Linux.

PDF export:
[image]

SVG-export:
[image]

jf---, 2019-02-06:
this is likely and issue with gl2ps, the library providing the conversion

cristobaltapia, 2019-02-07:
I was thinking that too. I'll post it there then.

cristobaltapia, 2019-02-08:
I have just run the tests included in `gl2ps` and they run without problem. I get the different vector formats saved as expected.
The problem must be somewhere else. I'm reopening this issue. 

SVG export:
[image]

PDF export:
[image]

tpaviot, 2019-02-12:
thanks for the report, I still have to test and confirm the issue

cristobaltapia, 2019-02-18:
If I can help you by providing any specific debugging information, just ask ;)