tpaviot/pythonocc-core #624: PackagesNotFoundError

Benatti1991, 2019-02-06:
Whenever I try to install PythonOCC with Anaconda I get this error:

```
conda install -c tpaviot pythonocc-core 
Solving environment: failed

PackagesNotFoundError: The following packages are not available from current channels:

  - pythonocc-core
  - oce==0.18.3
  - freeimageplus==3.17.0
  - pythonocc-core
  - smesh==6.7.6

Current channels:

  - https://conda.anaconda.org/tpaviot/linux-64
  - https://conda.anaconda.org/tpaviot/noarch
  - https://conda.anaconda.org/intel/linux-64
  - https://conda.anaconda.org/intel/noarch
  - https://conda.anaconda.org/conda-forge/linux-64
  - https://conda.anaconda.org/conda-forge/noarch
  - https://repo.anaconda.com/pkgs/main/linux-64
  - https://repo.anaconda.com/pkgs/main/noarch
  - https://repo.anaconda.com/pkgs/free/linux-64
  - https://repo.anaconda.com/pkgs/free/noarch
  - https://repo.anaconda.com/pkgs/r/linux-64
  - https://repo.anaconda.com/pkgs/r/noarch
  - https://repo.anaconda.com/pkgs/pro/linux-64
  - https://repo.anaconda.com/pkgs/pro/noarch

To search for alternate channels that may provide the conda package you're
looking for, navigate to

    https://anaconda.org

and use the search bar at the top of the page.
```
I never had any problem installing packages from channels

tpaviot, 2019-02-12:
Please carefully read the README. The proper way to install latest pythonocc-core release is

```
conda install -c conda-forge -c dlr-sc -c pythonocc -c oce pythonocc-core==0.18.1
```

If you wish to install nightly builds it's rather

```
conda install -c conda-forge -c dlr-sc -c pythonocc -c oce -c tpaviot pythonocc-core==0.18.2
```

Benatti1991, 2019-02-12:
Sorry, I figured out but i forgot to close the issue! Thank you, anyway!

closed 2019-02-12 by Benatti1991

tpaviot, 2019-02-12:
Can you please report how you solve the issue ?

Benatti1991, 2019-02-12:
Well, exactly as you mentioned (I found the instructions in the PythonOCC website): 
`conda install -c conda-forge -c dlr-sc -c pythonocc -c oce -c tpaviot pythonocc-core==0.18.2 `