tpaviot/pythonocc-core #627: Successfull compilation but libTKernel.so.11: cannot open shared object file on import

calocedrus, 2019-02-11:
I am installing python-occ on a linux box (ubuntu 18.04) and in a virtualenvironment(*). 

- Compiled and installed successfully OCE (in a folder under my /home directory).
- Compiled and installed (*apparently*) successfully python-occ the following way:
  - create a virtual environment for the install
  - cloned the latest branch with `git clone `
  - from a build directory, run `cmake` with the following options, specifying the location for the "non standard" location of OCE:
`cmake -DCMAKE_INSTALL_PREFIX=/home/ldv/.virtualenvs/cad_eg -DOCE_INCLUDE_PATH=/home/ldv/libs/OCE/include/oce -DOCE_LIB_PATH=/home/ldv/libs/OCE/lib ..`
  - run `make` (it takes a long time but compilation reports no error)
  - run `make install` (no need for `sudo` since the install target is in my user directory), all seems fine, with the last few lines from the install being:
 ```
[...]
-- Installing: /home/ldv/.virtualenvs/cad_eg/lib/python3.6/site-packages/OCC/XmlMXCAFDoc.py
-- Installing: /home/ldv/.virtualenvs/cad_eg/lib/python3.6/site-packages/OCC/XmlObjMgt.py
-- Installing: /home/ldv/.virtualenvs/cad_eg/lib/python3.6/site-packages/OCC/XmlTObjDrivers.py
-- Installing: /home/ldv/.virtualenvs/cad_eg/lib/python3.6/site-packages/OCC/XmlXCAFDrivers.py
-- Installing: /home/ldv/.virtualenvs/cad_eg/lib/python3.6/site-packages/OCC/gce.py
-- Installing: /home/ldv/.virtualenvs/cad_eg/lib/python3.6/site-packages/OCC/gp.py
-- Installing: /home/ldv/.virtualenvs/cad_eg/lib/python3.6/site-packages/OCC/math.py
```
In a python console I can do `import OCC`, no error.

But then when I try to import for example `OCC.IGESControl` from the same python console (with the virtual environment activated of course), I get:
```
File "/home/ludovic/.virtualenvs/cad_ge/lib/python3.6/site-packages/OCC/Core/IGESControl.py", line 25, in swig_import_helper
    return importlib.import_module(mname)
  File "/home/ludovic/.virtualenvs/cad_ge/lib/python3.6/importlib/__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 994, in _gcd_import
  File "<frozen importlib._bootstrap>", line 971, in _find_and_load
  File "<frozen importlib._bootstrap>", line 955, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 658, in _load_unlocked
  File "<frozen importlib._bootstrap>", line 571, in module_from_spec
  File "<frozen importlib._bootstrap_external>", line 922, in create_module
  File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
ImportError: libTKernel.so.11: cannot open shared object file: No such file or directory
```
But `libTKernel.so.11` is present in the correctly linked directory:
```
ls -al ~/libs/OCE/lib/libTKernel*
gonze:$ libTKernel.so -> libTKernel.so.11
gonze:$ libTKernel.so.11 -> libTKernel.so.11.0.0
gonze:$ libTKernel.so.11.0.0
```
And IGESControl.py does exist in the correct virtualenvs directory:
`-- Installing: /home/ldv/.virtualenvs/cad_eg/lib/python3.6/site-packages/OCC/IGESControl.py`

A similar errors has been reported in https://github.com/tpaviot/pythonocc-core/issues/563, but contrary to the author of the issue, I can't have my installation to work (it appears unclear how the problem was exactly solved, in particular what python flags were set). 

So: what can be the problem in my installation?

(*) I have started to use virtualenvironment to manage my python environments and it would to troublesome to switch to Anaconda, besides, some other python packages that I need are not available in Anaconda and others are not advised to install under anaconda.

tpaviot, 2019-02-11:
is ```/lib/OCE/lib``` in your ```LD_LIBRARY_PATH``` ? If no, try:

calocedrus, 2019-02-12:
I had done a `sudo ldconfig` after the installation of OCE and this obviously didn't do the trick given my local installation. I wasn't aware I had to add the path, but you are **right**! Many thanks!

So for any one facing similar errors, and/or wanting to install the great python-occ in a virtual environment, in addition to the steps above, one has to add the following line in `~/.bashrc`:

`export LD_LIBRARY_PATH=$LD_LIBRARY_PATH:/your/local/path/to/oce/lib`
And don't forget to source:
`source ~/.bashrc`

closed 2019-02-12 by calocedrus