tpaviot/pythonocc-core #628: BRepExtrema_ShapeProximity unusable?

tverho, 2019-02-12:
I'm trying to use `BRepExtrema_ShapeProximity` for quick intersection detection. I can run it, but the only way to get output are the functions `OverlapSubShapes1()` and `OverlapSubShapes2()`. Their return type is `BRepExtrema_MapOfIntegerPackedMapOfInteger`, which doesn't seem to have a proper wrapper. I cannot access any of its methods, even to see if it's empty or not.

`BRepExtrema_SelfIntersections` has a similar problem.

tpaviot, 2019-02-12:
can u please upload a ready-to-test sample code

tverho, 2019-02-12:
Sure, here you go:

```
from OCC.gp import gp_Pnt
from OCC.BRepPrimAPI import BRepPrimAPI_MakeBox
from OCC.BRepMesh import BRepMesh_IncrementalMesh
from OCC.BRepExtrema import BRepExtrema_ShapeProximity
 

box1 = BRepPrimAPI_MakeBox(gp_Pnt(0, 0, 0), gp_Pnt(2, 2, 2)).Shape()
box2 = BRepPrimAPI_MakeBox(gp_Pnt(1, 1, 1), gp_Pnt(3, 3, 3)).Shape()

# Create meshes for the proximity algorithm
deflection = 1e-3
mesher1 = BRepMesh_IncrementalMesh(box1, deflection)
mesher2 = BRepMesh_IncrementalMesh(box2, deflection)
mesher1.Perform()
mesher2.Perform()

tolerance = 0.1
isect_test = BRepExtrema_ShapeProximity(box1, box2, tolerance)
isect_test.Perform()

overlaps = isect_test.OverlapSubShapes1()

# overlaps doesn't have any methods like Size()
print(dir(overlaps))
```

tpaviot, 2019-02-12:
Confirmed.

tpaviot, 2019-12-29:
Same in latest 7.4.0beta2 release

tpaviot, 2019-12-29:
This works in 7.4.0beta2, but the ```NCollection_StlIterator``` class is not wrapped, and it's not possible to access faces that overlaps. See related issue #743

tpaviot, 2020-01-03:
Commit 4ae36a6aaf53c4f7af749b796ceca56466c2d727 fixes this issue.

I added an example core_geometry_overlap at https://github.com/tpaviot/pythonocc-demos/blob/master/examples/core_geometry_overlap.py (in red, faces that overlap, usefull to look for clashes)

[image]

tverho, 2020-01-03:
I can't test it because the latest master requires cmake 3.12 and my distro (Ubuntu 18.04 LTS) only has 3.10. However, I guess I don't need to test it to conclude that the bug is fixed and the issue can be closed. Nice work!

PR #749 (2020-01-05): Fixes in the wrapper and OCCViewer
* small cleanups in DisplayShape

* Added an extra ```Keys``` method to ```NCollection_DataMap``` template (fix #628)

* Remove unnecessary ```%nodefaultctor``` from many classes (fix #749)
closed 2020-01-03 by tverho