tpaviot/pythonocc-core #636: ModuleNotFoundError: No module named 'OCC' after succesfull installation

toxl, 2019-02-27:
Hi,

I have installed "pythonocc-core" via anaconda prompt by:
conda install -c conda-forge -c dlr-sc -c pythonocc -c oce pythonocc-core==0.18.1 python=3.6

Apparently, the installation was successful trusting the conda prompt message.

no if I run a simple example like this one:
https://github.com/tpaviot/pythonocc-demos/blob/master/examples/core_animation.py
also following the hint to modify OCC.core.xxx to OCC.xxx from here:
https://github.com/tpaviot/pythonocc-core/issues/565

I get always a “ModuleNotFoundError: No module named 'OCC'”!

Any idea how to get this to work?

Thanks for your help!

toxl, 2019-02-28:
Hi,

Problem solved.
the local spyder installation was not running in the same environment (base) where pythonocc-core was installed.
Once this was solved it worked immediately.

Sorry for the noise!

tpaviot, 2019-03-01:
No problem, thank you for the report anyway. I changed the issue title to add "Spyder", in the case someone else faces the same issue.

closed 2019-03-01 by tpaviot

volkan1986, 2022-01-10:
Hello i couldn't understand how the problem solved. Please help me 

I get always  “ModuleNotFoundError: No module named 'OCC'”! 

how can i fix it please help

tpaviot, 2022-01-11:
You have to be sure that you installed the right pythonocc version/python version. Which version did you install ? which os ?

volkan1986, 2022-01-11:
Thank you very much for replying.
I have installed "pythonocc-core" via anaconda prompt by: conda create -n pyoccenv python=3.9.7
than i have written; conda activate pyoccenv
and than; conda install -c conda-forge pythonocc-core.

Meanwhile i can the code by using anaconda by entering the below code.
C:/users/Volkan/desktop/3dcopy.py

But while i try to run it via visual studio, i get "ModuleNotFoundError: No module named 'OCC'"
How can I fix it?

Thank you very much for your interest, sorry for my inexperience.

yu9731, 2022-09-06:
Hello, I have installed pythonocc-core sucessfully via cmake in linux. But when I ran the run_test.py, I always got the same error as "ModuleNotFoundError: No module named 'OCC". I want to know, what's wrong with it?