tpaviot/pythonocc-core #637: Pythonocc-core 0.18.1 breaks conda command in Conda virtual enviroment

Durksz, 2019-02-27:
I'm trying to create an executable of my program that makes use of pythonocc. To ensure that the executable is as clean as possible i create the executable within a virtual enviroment. I create this virtual enviroment using conda (conda create --name envname). 
When I try to install pythonocc using the default installation line found in pythonocc.org "conda install -c conda-forge -c dlr-sc -c pythonocc -c oce pythonocc-core==0.18.1" the installation succeeds. But when trying to acces conda, for example "conda list" conda is not found anymore. 
I found this to be caused by the OCE version 0.18.2. Because when I install only oce=0.18.2, the same problem occurs. 
When installing oce=0.18.1 or oce=0.18.3 conda keeps working correctly. 

My workaround at this moment is to install pythonocc-core==0.18.0 this is dependent on oce=0.18.1 in stead of oce=0.18.2.

A solution might be to update the pythonocc=0.18.1 to be dependent on oce=0.18.3

tpaviot, 2019-03-01:
I'm aware of this issue.  It should be fixed in the current master branch. Can you please tell me if the following works

```
conda install -c conda-forge -c dlr-sc -c tpaviot -c oce pythonocc-core==0.18.2
``

Durksz, 2019-03-01:
Conda does not seem to find the 0.18.2 version

tpaviot, 2019-03-02:
you're on Windows, the 'main' label was not set on anaconda. It's fixed, the package should be found now.

closed 2019-03-02 by tpaviot

Durksz, 2019-03-04:
It works, thanks!