tpaviot/pythonocc-core #638: cartesian points extraction of selected face fails

toxl, 2019-02-28:
Hi tpaviot,

I would like to extract coordinates of a single surface of a step-file model (file is attached below) and screenshot of the desired surface can be see below too.
I would like to select the surface via mouse and extract the data points into a variable. 
I started of with code taken from:
https://github.com/tpaviot/pythonocc-demos/blob/master/examples/core_geometry_face_recognition_from_stepfile.py
for the face recognition which works fine after some modification (code is attached).

However, I am not able to modify this example:
https://github.com/tpaviot/pythonocc-demos/blob/master/examples/core_topology_uv_to_cartesian_coordinates.py
to extract all points associated with the selected surface.
Would you be able to support me in order to extract the desired data? I am happy to contribute the stp-file and all the code to the example repository.

[image]

[example_model.stp.txt](https://github.com/tpaviot/pythonocc-core/files/2916058/example_model.stp.txt)
[core_geometry_face_recognition_from_stepfile.py.txt](https://github.com/tpaviot/pythonocc-core/files/2916082/core_geometry_face_recognition_from_stepfile.py.txt)

Cheers

tpaviot, 2019-03-02:
Everything is available in those two scripts to perform what is expected. It seems the selected face is a plane. Make a face from the selected plane using ```BRepBuilderAPI_MakeFace``` and use the code from core_topology_uv_to_cartesian_coordinates.py

closed 2019-03-02 by tpaviot

toxl, 2019-03-04:
Hi,

thanks for the reinforcement! I am not quite sure how to apply `BRepBuilderAPI_MakeFace` since what comes out of `def recognize_clicked()` is already a face instance in my case. 
In this case the input into the function is already of type `TopoDS_Face`. Now it is rather that `ShapeAnalysis_Surface()` needs a `Handle_Geom_Surface` as argument. And to get `Handle_Geom_Surface()` from "TopoDS_shape" the method `Surface()` needs to be applied twice like `ShapeAnalysis_Surface(surf.Surface().Surface())` (see attached code line 106)).
Please correct me if I am wrong.

Once this was solved, the example code was modified such that at the end of the routine a lists of Cartesian [x,y,z] data points were stored in a file.

Reloading the extracted data and plotting them via `matplotlib.pyplot.scatter()` revealed a square plane filled with plenty of XYZ points in a different orientation!?! (see image)
[image]

I try to be clearer: I am looking for the ability to extract only but exactly the edge of the selected plane. Is there a method such as `shapeanalysis_GetEdgeFromFace` or alike? (The documentation was not very helpful in this case).

Thanks again for your valued support!
[20190304_stpFileRead_surface2CartesianCoordinates.py.txt](https://github.com/tpaviot/pythonocc-core/files/2926667/20190304_stpFileRead_surface2CartesianCoordinates.py.txt)