tpaviot/pythonocc-core #640: Problem running example core_visualization_overpaint_viewer.py (pythonocc v0.18.2)

carlosantonio89, 2019-03-03:
Hi guys, 

I've just installed the latest version of pythonocc (v0.18.2) using conda environment.  Everything was fine and many examples worked perfectly. When I tried to run the "core_visualization_overpaint_viewer.py" (file attached) I got the following error:

`QWindowsGLContext::makeCurrent: SetPixelFormat() failed (The pixel format is invalid.)`

And the window closes. I am running the code on Windows 10, 64 bits, Nvidia card, Python 3.6.8 (Anaconda Package). 

This example is very important for the application I want to use. I appreciate all help. Thanks in advance.

[core_visualization_overpaint_viewer.zip](https://github.com/tpaviot/pythonocc-core/files/2923660/core_visualization_overpaint_viewer.zip)

carlosantonio89, 2019-03-03:
One more thing: Running other examples as: core_visualization_graphic3d_custom_opengl.py and core_visualization_3d_to_2d_screen_coordinates.py I replicated the issue. These examples work nice but when I press "Shift key" and the any mouse button the program crashes.

tpaviot, 2019-03-05:
I cannot reproduce the issue. Which GUI manager do you use ?

The ```SetPixelFormat()``` issue is usually raised in case there is an error with the OpenGL driver. Be sure your graphic driver is up to date.

carlosantonio89, 2019-03-05:
Thank you for your reply Thomas.

I am using PyQt 5.9.2. My video card is on it's latest driver version:

**Video card:** Geforce GT 540M/PCIe/SSE2 (**Driver version:** 391.35)
**OpenGL version:** 4.6

Any idea of such error under this circumstances?

jf---, 2019-03-11:
@carlosantonio89 which version of osx are you on? I'll see if I can replicate