tpaviot/pythonocc-core #641: BRepAlgoAPI_Fuse fails

sfornengo, 2019-03-06:
the following code fuses 2 torus then a cylinder and fails...
if using another cylinder (please uncomment in code), it works:

`from OCC.BRepAlgoAPI import BRepAlgoAPI_Fuse
from OCC.BRepBuilderAPI import BRepBuilderAPI_Transform
from OCC.gp import gp_Vec, gp_Trsf
from OCC.BRepPrimAPI import BRepPrimAPI_MakeCylinder, BRepPrimAPI_MakeTorus
from OCC.gp import gp_Trsf
from OCC.Display.SimpleGui import init_display

import math

display, start_display, add_menu, add_function_to_menu = init_display(size=(1500,1000))
display.EraseAll()

def m2trsf(m):
    trsf = gp_Trsf()
    trsf.SetValues(*m)
    return trsf

    
d={'name': 'circulartorus', 'attrs': {}, 'dim': {'smallradius': 330.0, 'largeradius': 370.0, 'angle': 180.0}, 'm': [0.9781, 0.0, 0.2079, -12722.8925, -0.2079, 0.0, 0.9781, 2113.0701, 0.0, -1.0, 0.0, 48182.0]}
dim=d['dim']
torus1 = BRepPrimAPI_MakeTorus((dim['largeradius']+dim['smallradius'])/2., (dim['largeradius']-dim['smallradius'])/2., dim['angle']*math.pi/180)
torus1=BRepBuilderAPI_Transform(torus1.Shape(), m2trsf(d['m']), False)

d={'name': 'circulartorus', 'attrs': {}, 'dim': {'smallradius': 330.0, 'largeradius': 370.0, 'angle': 180.0}, 'm': [0.9781, 0.0, -0.2079, -12722.8925, -0.2079, 0.0, -0.9781, 2113.0701, 0.0, 1.0, 0.0, 48182.0]}
dim=d['dim']
torus2 = BRepPrimAPI_MakeTorus((dim['largeradius']+dim['smallradius'])/2., (dim['largeradius']-dim['smallradius'])/2., dim['angle']*math.pi/180)
torus2=BRepBuilderAPI_Transform(torus2.Shape(), m2trsf(d['m']), False)

f1=BRepAlgoAPI_Fuse(torus1.Shape(),torus2.Shape())

d={'name': 'cylinder', 'attrs': {}, 'dim': {'radius': 20.0, 'height': 700.0}, 'm': [0.2079, 0.9781, 0.0, -12722.8925, 0.9781, -0.2079, 0.0, 2113.0701, 0.0, 0.0, -1.0, 48182.0]}
#d={'name': 'cylinder', 'attrs': {}, 'dim': {'radius': 20.0, 'height': 700.0}, 'm': [0.0, 0.2079, -0.9781, -12722.8925, 0.0, 0.9781, 0.2079, 2113.0701, 1.0, 0.0, 0.0, 48182.0]}
dim=d['dim']
cyl = BRepPrimAPI_MakeCylinder(dim['radius'],dim['height'])
trsf = gp_Trsf()
trsf.SetTranslation(gp_Vec(0.,0.,-dim['height']/2.))
cyl=BRepBuilderAPI_Transform(cyl.Shape(), trsf, False)
cyl=BRepBuilderAPI_Transform(cyl.Shape(), m2trsf(d['m']), False)

f2=BRepAlgoAPI_Fuse(f1.Shape(),cyl.Shape())

display.DisplayShape(f2.Shape(), update=True)
start_display()`

sfornengo, 2019-03-07:
I am using python-occ 0.18.1 installed with conda/python 3.6.7.
It seems the use case above fails when the cylinder extremities are on the 2 torus fusion join...
Any idea to get around of the problem ?

tpaviot, 2019-03-17:
Can you please post the simplest example that reproduces the problem (remove dicts, oerations on diameters etc.). It will be easier to debug.

sfornengo, 2019-03-18:
Here is a simplified version:
The simplification let shown that the bug can be triggered changing the x-translation part tx, just uncomment to trigger the bug:
`#!/usr/bin/env python
# -*- coding: utf-8 -*-

from OCC.BRepAlgoAPI import BRepAlgoAPI_Fuse
from OCC.BRepBuilderAPI import BRepBuilderAPI_Transform
from OCC.gp import gp_Vec, gp_Trsf
from OCC.BRepPrimAPI import BRepPrimAPI_MakeCylinder, BRepPrimAPI_MakeTorus
from OCC.gp import gp_Trsf
from OCC.Display.SimpleGui import init_display

import math

display, start_display, add_menu, add_function_to_menu = init_display()
display.EraseAll()

tx=0. # no bug
#tx=-12722.8925 # bug
ty=2113.0701
tz=48182.0

torus1 = BRepPrimAPI_MakeTorus(350., 20., 180*math.pi/180)
trsf = gp_Trsf()
trsf.SetValues(0.9781, 0.0, 0.2079, tx, -0.2079, 0.0, 0.9781, ty, 0.0, -1.0, 0.0, tz)
torus1=BRepBuilderAPI_Transform(torus1.Shape(), trsf, False)

torus2 = BRepPrimAPI_MakeTorus(350., 20., 180*math.pi/180)
trsf = gp_Trsf()
trsf.SetValues(0.9781, 0.0, -0.2079, tx, -0.2079, 0.0, -0.9781, ty, 0.0, 1.0, 0.0, tz)
torus2=BRepBuilderAPI_Transform(torus2.Shape(), trsf, False)

f1=BRepAlgoAPI_Fuse(torus1.Shape(),torus2.Shape())

cyl = BRepPrimAPI_MakeCylinder(20.,700.)
trsf = gp_Trsf()
trsf.SetTranslation(gp_Vec(0.,0.,-700./2.))
cyl=BRepBuilderAPI_Transform(cyl.Shape(), trsf, False)
trsf = gp_Trsf()
trsf.SetValues(0.0, 0.2079, -0.9781, tx, 0.0, 0.9781, 0.2079, ty, 1.0, 0.0, 0.0, tz)
cyl=BRepBuilderAPI_Transform(cyl.Shape(), trsf, False)

f2=BRepAlgoAPI_Fuse(f1.Shape(),cyl.Shape())

display.DisplayShape(f2.Shape(), update=True)
start_display()`

sfornengo, 2019-03-18:
tx is just the x-position of each parts in the global space, it would normally not impact something...

tpaviot, 2019-03-19:
You can perform the second fuse operation using a fuzzy parameter:
```python
from OCC.Core.TopTools import TopTools_ListOfShape
    
f2 = BRepAlgoAPI_Fuse()
L1 = TopTools_ListOfShape()
L1.Append(f1.Shape())
L2 = TopTools_ListOfShape()
L2.Append(cyl.Shape())
f2.SetArguments(L1)
f2.SetTools(L2)
f2.SetFuzzyValue(0.01)
f2.Build()
```
it works on my machine, please test and close the issue if the result suits you.

sfornengo, 2019-03-19:
Thank you very much, the fuse is now performed.
Could you quickly explain or point to the documentation about this parameter please ?

closed 2019-03-19 by sfornengo

tpaviot, 2019-03-19:
Fuzzy logic boolean operations https://www.opencascade.com/doc/occt-6.9.1/overview/html/occt_user_guides__boolean_operations.html#occt_algorithms_11a_1

sumsbee, 2020-09-24:
I dont't know if this is the right place to ask but I didn't find anything else.

I am using pythonocc-core 7.4.0 installed with conda python 3.7.8.

The method .Build() dosn't do anything and as far as I understand the inline doc it is also not intended to do so. I can set the FuzzyValue but I don't manage to start the operation. The code above dose not calculate the fuse. Am I mixing up things?

Most examples I find use the fuse algorithm without using the fuzzy value.