tpaviot/pythonocc-core #642: DepreciationWarning: 'input_splitter' is deprecieated since IPython 7.0 prefer 'input_transformer_manager'

Durksz, 2019-03-11:
I installed Pythonocc 0.18.2 on my windows 7 64bit machine.

When i try to run the following code:
`from OCC.wxDisplay import wxViewer3D`

I get the following warning:
C:\ProgramData\Miniconda3\lib\site-packages\ipykernel\ipkernel.py:448: DeprecationWarning: `input_splitter` is deprecated since IPython 7.0, prefer `input_transformer_manager`.
  status, indent_spaces = self.shell.input_splitter.check_complete(code)

jf---, 2019-03-11:
The message is pretty clear: `ipykernel` ( != `pythonocc` ) has a deprecation 

> `from OCC.wxDisplay import wxViewer3D`

Yep, it would be particular to have ipykernel msg's when importing the `wx` backend.
So this is coming from somewhere else in your program p'bably

Durksz, 2019-03-13:
Hmm, Your right. Sorry my fault.

tpaviot, 2019-03-17:
Something that certainly comes from wxPython itself, or ipython, or both. According to https://github.com/spyder-ide/spyder/issues/8103, you should try to downgrade to ipython6. @Durksz Can you please report any success/failure so that we can safely close this issue.

Durksz, 2019-03-18:
I've downgraded PythonOCC to 0.18.1 and the error still occurs, definately not an PythonOCC problem. Have not found the specific package (and its version) yet that causes the problem.

Durksz, 2019-03-18:
After upgrading all the packages using "conda update --all" (causing PythonOCC to downgrade to 0.17.3, so I followed by the command "conda install PythonOCC-Core==0.18.2") the problem does not occur anymore.

tpaviot, 2019-03-18:
ok @Durksz  I suggest you close this issue

Durksz, 2019-03-18:
Closed #642

PCdurham, 2019-07-18:
I updated conda yesterday and with Spyder 3.3.6 this issue still occurs.  It's quite serious since the iPython console is rendered useless.

ReedEspinosa, 2020-03-25:
In my case, calling `warnings.simplefilter('always')` is what causes the warning to start appearing. Note that this line can be found multiple times in [OCC](https://github.com/tpaviot/pythonocc-core).

To reproduce the following example, each input must be copy/pasted into the console and executed one at a time:
```
In [1]: import warnings

In [2]: 2+2 # no warning message
Out[2]: 4

In [3]: warnings.simplefilter('always')

In [4]: 2+2 # displays warning message
/Users/wrespino/anaconda3/lib/python3.6/site-packages/ipykernel/ipkernel.py:454: DeprecationWarning: `input_splitter` is deprecated since IPython 7.0, prefer `input_transformer_manager`.
  status, indent_spaces = self.shell.input_splitter.check_complete(code)
Out[4]: 4

In [5]: warnings.filters.pop(0) # remove ('always', None, Warning, None, 0) from start of warnings.filters
/Users/wrespino/anaconda3/lib/python3.6/site-packages/ipykernel/ipkernel.py:454: DeprecationWarning: `input_splitter` is deprecated since IPython 7.0, prefer `input_transformer_manager`.
  status, indent_spaces = self.shell.input_splitter.check_complete(code)
Out[5]: ('always', None, Warning, None, 0)

In [6]: 2+2 # no warning message
Out[6]: 4
```

My setup:
`Spyder 4.1.0 | IPython 7.13.0 | Python 3.6.8 64-bit | Qt 5.9.6 | PyQt5 5.9.2 | Darwin 18.7.0`