tpaviot/pythonocc-core #645: webgl_threejs can't display model correctly.

liuxin2322, 2019-03-16:
in web brower,model can't be displayed correctly. some surface are missing.
[image]
i confirm the model is right because it was tested and displayed entirely in Qt.

[image]

source code :
```python
from OCC.Extend.DataExchange import read_iges_file
from OCC.Display.WebGl import threejs_renderer

big_shp = read_iges_file("j3.igs")

my_renderer = threejs_renderer.ThreejsRenderer()
my_renderer.DisplayShape(big_shp,color=(1,1,0))
my_renderer.render()

```

tpaviot, 2019-03-17:
I need the igs file to test

liuxin2322, 2019-03-17:
> I need the igs file to test

ok,here. thank you in advance

[j3.zip](https://github.com/tpaviot/pythonocc-core/files/2974850/j3.zip)

tpaviot, 2019-03-17:
Issue confirmed.

tpaviot, 2019-03-17:
which os do you run ?

liuxin2322, 2019-03-17:
> which os do you run ?

win10，pythonocc0.18.2

tpaviot, 2019-03-17:
The issue seems to be related to the threejs renderer only. On my machine (Ubuntu 18.04), the x3dom renderer works as expected. Can you please try

liuxin2322, 2019-03-17:
> The issue seems to be related to the threejs renderer only. On my machine (Ubuntu 18.04), the x3dom renderer works as expected. Can you please try

you're right,x3dom works well.  is there any solutions about threejs?

tpaviot, 2019-03-17:
I got the threejs renderer working by forcing double sided rendering. You can force double siding by modifying the index.html file, and adding the property ```side: THREE.DoubleSide``` to the MeshPhongMaterial.

closed 2019-03-18 by liuxin2322

tpaviot, 2019-03-18:
Fixed by commit d6a4623532182bd9cdfc84455b690e3f458e91b4