tpaviot/pythonocc-core #648: Anaconda tries to force downgrade to python 2.7.15 upon python-occ install

jrj99, 2019-04-12:
Hi, I am trying to install pythonOCC using the anaconda prompt. when i run this command in my python 3.6 environment:

conda install -c conda-forge -c dlr-sc -c pythonocc -c oce pythonocc-core==0.18.1

Conda always tries to force the environment to downgrade to 2.7. I am using miniconda3.

Here is my error log:

(pythonOCC) H:\Projects\pyOCC_mini_project(new)\pyOCC_component>conda install -c
 conda-forge -c dlr-sc -c pythonocc -c oce pythonocc-core==0.18.1
Solving environment: done

==> WARNING: A newer version of conda exists. <==
  current version: 4.5.12
  latest version: 4.6.11

Please update conda by running

    $ conda update -n base -c defaults conda

## Package Plan ##

  environment location: C:\Users\caelab\Miniconda3\envs\pythonOCC

  added / updated specs:
    - pythonocc-core==0.18.1

The following NEW packages will be INSTALLED:

    ca-certificates: 2019.3.9-hecc5488_0      conda-forge
    freeimageplus:   3.17.0-vc9_0             dlr-sc      [vc9 ]
    freetype:        2.7-vc9_1                conda-forge
    gl2ps:           1.3.8-vc9_0              dlr-sc      [vc9 ]
    icu:             58.2-vc9_0               conda-forge
    jpeg:            9c-h0c8e037_1001         conda-forge
    libpng:          1.6.36-h7a46e7a_1000     conda-forge
    oce:             0.18.2-vc9hfba03eb_1     oce         [vc9 ]
    openssl:         1.0.2r-h0c8e037_0        conda-forge
    pyqt:            5.6.0-py27h4cbc711_1008  conda-forge
    qt:              5.6.2-vc9_1              conda-forge
    sip:             4.18.1-py27hc56fc5f_1000 conda-forge
    smesh:           6.7.5-vc9ha7c2d4a_0      pythonocc   [vc9 ]
    tbb:             2017_20170226-vc9_0      conda-forge
    vs2008_runtime:  9.0.30729.6161-0         conda-forge
    zlib:            1.2.11-h3cc03e0_1004     conda-forge

The following packages will be UPDATED:

    certifi:         2019.3.9-py36_0                      --> 2019.3.9-py27_0
   conda-forge
    pip:             19.0.3-py36_0                        --> 19.0.3-py27_0
   conda-forge
    pythonocc-core:  0.18.1-py36_vc14_110     <unknown>   [vc14] --> 0.18.1-py27
_vc9_110  pythonocc   [vc9]
    setuptools:      41.0.0-py36_0                        --> 41.0.0-py27_0
   conda-forge
    wheel:           0.33.1-py36_0                        --> 0.33.1-py27_0
   conda-forge
    wincertstore:    0.2-py36h7fe50ca_0                   --> 0.2-py27_1002
   conda-forge

The following packages will be DOWNGRADED:

    python:          3.6.8-h9f7ef89_7                     --> 2.7.15-h2880e7c_10
08 conda-forge
    sqlite:          3.27.2-he774522_0                    --> 3.26.0-h0c8e037_10
01 conda-forge
    vc:              14.1-h0510ff6_4                      --> 9-0
   conda-forge

TristanDoherty, 2019-05-15:
Hi jr99, 

I had similar issues and it seems the dlr-sc channel doesn't include some of the dependencies for python 3.6. Try adding the 3dhubs channel:

conda install -c conda-forge -c tpaviot -c 3dhubs -c pythonocc -c dlr-sc -c oce pythonocc-core=0.18.1 python=3.6

pparkes, 2019-05-25:
worked for me, thanks!

vaskokj, 2019-12-19:
This is an issue for me as well.

ran `conda install -c conda-forge -c tpaviot -c 3dhubs -c pythonocc -c dlr-sc -c oce pythonocc-core=0.18.1 python=3.6` oddly enough it forces a downgrade to python-3.6.7 even though 3.6.9 is available (which I had already installed).

tpaviot, 2020-01-05:
Try install the latest version

```
conda create --name=whatever-name python=3.7
source activate whatever-name
conda install -c dlr-sc -c pythonocc pythonocc-core=7.4.0rc1
````

and reopen the issue if it still fails.

closed 2020-01-05 by tpaviot