tpaviot/pythonocc-core #649: issues when trying to install v0.18.x under Python 3.6 or 3.7

daniel1v, 2019-04-13:
I tried different methods to install pythonocc-core under python 3.6 and 3.7. Every method failed.

**1)**
`conda install -c conda-forge -c dlr-sc -c pythonocc -c oce pythonocc-core==0.18.1 python=3.6`
leads to:
```
Solving environment: failed

UnsatisfiableError: The following specifications were found to be in conflict:
  - python=3.6
  - pythonocc-core==0.18.1
Use "conda search <package> --info" to see the dependencies for each package.
```

**2)**
`conda install -c tpaviot pythonocc-core `
leads to a downgrade to python 2.7

**3)**
`conda install -c tpaviot/label/cd-review/prepare-0.18.2 pythonocc-core `
leads to 
```
UnavailableInvalidChannel: The channel is not accessible or is invalid.
  channel name: tpaviot/label/cd-review/prepare-0.18.2
  channel url: https://conda.anaconda.org/tpaviot/label/cd-review/prepare-0.18.2
  error code: 404

You will need to adjust your conda configuration to proceed.
Use `conda config --show channels` to view your configuration's current state,
and use `conda config --show-sources` to view config file locations.
```

**4)**
`conda install -c tpaviot/label/cd-master pythonocc-core`
leads to a downgrade to python 2.7

I also tried to vary the version number between 0.18.1 and 0.18.2 for every method and the python version between 3.6 and 3.7. Nothing worked for me. I tried it on two different win 10 64bit machines. 

In the past I hand no problems installing pythonocc with conda. So I think there must be something wrong with the conda packages.

Could anyone double check and maybe provide support?

daniel1v, 2019-04-14:
addition: 
the 0.18.2 version from the cadquery channel installed without any issue
(https://anaconda.org/CadQuery/pythonocc-core)

jrj99, 2019-04-15:
I have been having the same issue, and I created an Issue on this subject yesterday as well.

I even tried specifying the version number with this command:

conda install -c conda-forge -c dlr-sc -c python occ -c oce pythonocc-core==0.18.1=py36_vc14_110

which resulted in the error code:

(pythonOCC) C:\Windows\system32>conda install -c conda-forge -c dlr-sc -c python
occ -c oce pythonocc-core==0.18.1=py36_vc14_110

which results in this error:

Collecting package metadata: done
Solving environment: failed

UnsatisfiableError: The following specifications were found to be in conflict:
  - pythonocc-core==0.18.1=py36_vc14_110
Use "conda search <package> --info" to see the dependencies for each package.

wzbernstein, 2019-05-02:
I have had similar issues with 0.18.  I am trying to update a build from 0.17.  When I update to 0.18 in my environment, the prompt can no longer find the path to conda/python.   A simple command such as "conda list" returns "The system cannot find the path specified."  I ran many tests to no avail.  I even conducted a fresh install of conda and the same problem persisted.  I checked if the old env (with 0.17) still worked and it did.

waterbug, 2019-05-02:
I had problems like you describe when using 0.17.  I installed 0.18.2 from the cadquery channel in a python 3.6 env and have not had problems since.

tpaviot, 2019-05-02:
Warning : pythonocc-0.18.2 is not released yet. The latest pythonocc release is 0.18.1. I don't know wich github tag is targetted by the cadquery anaconda channel. I recommend using the https://anaconda.org/tpaviot/pythonocc-core channel if you want to download/installed nightly binaries

waterbug, 2019-05-02:
Sorry but 0.18.1 is broken.

wzbernstein, 2019-05-02:
I tried the .../tpaviot/pyhonocc-core options, but each one returned an error.  I will try with cadquery and see what happens.

wzbernstein, 2019-05-22:
> Warning : pythonocc-0.18.2 is not released yet. The latest pythonocc release is 0.18.1. I don't know wich github tag is targetted by the cadquery anaconda channel. I recommend using the https://anaconda.org/tpaviot/pythonocc-core channel if you want to download/installed nightly binaries

We noticed that you the pythonocc-core channel was updated today.  I tried to install the new version but got the following error:

PackagesNotFoundError: The following packages are not avilable from current channels:
- pythonocc-core -> oce==0.18.3
- pythonocc-core -> smesh==6.7.6

chenming-wu, 2019-06-13:
I think the `cadquery` channel is not complete, I tried the following command and it works well on my machine.
```
conda install -c tpaviot -c conda-forge -c dlr-sc -c oce -c pythonocc pythonocc-core=0.18.2
```

waterbug, 2019-06-13:
Thanks so much for adding that useful comment!  The set of channels you specify (which excludes the cadquery channel) also works for me, in both a python 3.6 environment and a 3.7 environment on linux. My earlier comment should be disregarded -- apologies if it caused any confusion!

frincyc, 2019-06-19:
> I think the `cadquery` channel is not complete, I tried the following command and it works well on my machine.
> 
> ```
> conda install -c tpaviot -c conda-forge -c dlr-sc -c oce -c pythonocc pythonocc-core=0.18.2
> ```

This works well on my machine too, but python is again downgraded from 3.5.6 to 2.7.15. 
Afterwhich, when I run some example tutorials, scipy/numpy version supported by python 2.7 version returns errors that few modules are not found. How can we have both python 3.5+ and pythonocc installed ?

chenming-wu, 2019-06-19:
> This works well on my machine too, but python is again downgraded from 3.5.6 to 2.7.15.
> Afterwhich, when I run some example tutorials, scipy/numpy version supported by python 2.7 version returns [...]

I'm using Python 3.6 and it works well for me. Can you try this?

frincyc, 2019-06-19:
[image]
It still says the same..
 python                 pkgs/main::python-3.6.5-h0c2934d_0 --> conda-forge::python-2.7.15-h2880e7c_1008

frincyc, 2019-06-19:
> [image]
> It still says the same..
> python pkgs/main::python-3.6.5-h0c2934d_0 --> conda-forge::python-2.7.15-h2880e7c_1008

> > > I think the `cadquery` channel is not complete, I tried the following command and it works well on my machine.
> > > ```
> > > conda install -c tpaviot -c conda-forge -c dlr-sc -c oce -c pythonocc pythonocc-core=0.18.2
> > > ```
> > 
> > 
> > This works well on my machine too, but python is again downgraded from 3.5.6 to 2.7.15.
> > Afterwhich, when I run some example tutorials, scipy/numpy version supported by python 2.7 version returns errors that few modules are not found. How can we have both python 3.5+ and pythonocc installed ?
> 
> I'm using Python 3.6.5 and it works well for me. Can you try this?

Pls see my response above...

glinskyc, 2019-06-19:
This is related to [issue 648](https://github.com/tpaviot/pythonocc-core/issues/648).

Adding `3dhubs` to channels should resolve it.  So your install command should look like this in the case of Python 3.6:

```
conda install -c conda-forge -c dlr-sc -c pythonocc -c oce -c 3dhubs pythonocc-core==0.18.1 python=3.6
```

I expect it will also work for Python 3.7 but I have not tried.

frincyc, 2019-06-19:
> This is related to [issue 648](https://github.com/tpaviot/pythonocc-core/issues/648).
>
> Adding `3dhubs` to channels should resolve it. So your install command should look like this in the case of [...]

Hey thank you @glinskyc ! It worked!
I installed on an env with python 3.6.8 and above install command, it downgraded python to 3.6.7, but I was able to get it up and running! Thanks

glinskyc, 2019-06-25:
Has anyone figured out how to get `pythonocc-core==0.18.1` to install on Linux?  Only `0.18.2` is working.

glinskyc, 2019-06-25:
On a Linux 64-bit system, the following is encountered when attempting the install of `python-occ==0.18.1`:

```
UnsatisfiableError: The following specifications were found to be in conflict:
  - pythonocc-core=0.18.1
Use "conda search <package> --info" to see the dependencies for each package.
```
The results from `conda search pythonocc-core==0.18.1 --info -c pythonocc` is the following for linux-64 py36 build:

```
pythonocc-core 0.18.1 py36hdfe9f0f_110
--------------------------------------
file name   : pythonocc-core-0.18.1-py36hdfe9f0f_110.tar.bz2
name        : pythonocc-core
version     : 0.18.1
build       : py36hdfe9f0f_110
build number: 110
size        : 47.9 MB
license     : LGPL
subdir      : linux-64
url         : https://conda.anaconda.org/pythonocc/linux-64/pythonocc-core-0.18.1-py36hdfe9f0f_110.tar.bz2
md5         : a7876a3300e2b58b80ddbc313f9fed4d
timestamp   : 2017-12-14 06:59:41 UTC
dependencies:
  - freetype ==2.7
  - oce ==0.18.2
  - pyqt >=5
  - python >=3.6,<3.7.0a0
  - smesh ==6.7.5
```
I think the issue may lie within the dependencies, which are unfortunately the same for each build, except the python version.  

Any help with figuring out what the issue is would be appreciated.

sfornengo, 2019-07-15:
This is working for me:
conda install -c tpaviot -c conda-forge -c dlr-sc -c pythonocc -c oce -c 3dhubs pythonocc-core==0.18.2 python=3.7

jrj99, 2019-08-02:
The official pythonOCC cloud install is still broken, any ETA on when this will get fixed?

wangjiawen2013, 2019-08-04:
I met the same problem. According to my experience, pythonocc0.18 is only compatible with python2, while pythonocc 0.17 is compatible with python3. We cannot run pythonocc 0.18 in python3 environment. @tpaviot

Durksz, 2019-08-27:
I tried to install the pythonocc-core (0.18.2) yesterday, and had the same problem as stated above. This while it worked I installed pythonocc-core = 0.18.2  the last time (in march 2019). The code from sfornengo solved this problem. Apparently some of the dependencies of Pythonocc-core have moved from their original location.

theveloped, 2019-10-08:
I'm still experiencing similar error when updating one of our legacy projects running on python 2.7 and pythonocc-core 0.18.1. This following command used to work:
```
conda install -c tpaviot -c conda-forge -c dlr-sc -c pythonocc -c oce pythonocc-core==0.18.1 python=2
```
Adding the `3dhubs` channel or specifying the python version according to the new docs as `python=2.7` doesn't seem to have any effect either. The error returned is:
```
UnsatisfiableError: The following specifications were found to be incompatible with a past
explicit spec that is not an explicit spec in this operation (python):

  - pythonocc-core==0.18.1 -> python[version='>=2.7,<2.8.0a0|>=3.5,<3.6.0a0|>=3.6,<3.7.0a0']
```
Any idea's on how to get this legacy version up and running while we upgrade to python 3 and pythonocc-core 0.18.2?

tpaviot, 2020-01-05:
Please move to latest release.

closed 2020-01-05 by tpaviot