tpaviot/pythonocc-core #650: OCC.BRepClass.BRepClass_FaceClassifier does not get wrapped

sdh4, 2019-04-19:
The 3D face classifier OCC.BRepClass.BRepClass_FaceClassifier is not currently wrapped by the bindings. Its superclass OCC.BRepClass.BRepClass_FClassifier that does parametric point classification does get wrapped by the bindings.

tpaviot, 2019-04-22:
Confirmed.

Fixed by commit 3a77a98d

lambrtz, 2020-10-23:
Hello, sorry for necro-ing an old post, but I cannot seem to find BRepClass_FaceClassifier in miniconda, channel conda-forge. Am I missing something?

lambrtz, 2020-10-24:
I just noticed that BRepClass_FaceClassifier disappears again after this commit: ce671d360676aa3750c856c4d8252383bc897528 
Is it possible that it is re-introduced to pythonocc?

tpaviot, 2020-11-06:
done, see commit 902279387bc58152d4c6f8b46ee271f1efb46098

closed 2020-11-06 by tpaviot