tpaviot/pythonocc-core #651: bridge between pythonocc and vtk

sjtu-xx, 2019-04-23:
I'm trying to use pythonocc reading a .stp file, and display through VTK. 
But when I try to import OCC.Core.IVtkOCC, the program raise ModuleNotFoundError error.
I'm using pythonocc 0.18.2. what should I do to fix it?
thanks!

tpaviot, 2019-04-27:
The VTK part of oce is not wrapped by pythonocc, and there's not plan for such a support

closed 2019-04-27 by tpaviot

insert-username, 2022-11-11:
Not sure if this is the best place to post this, but...

Commenting for anyone else searching for a workaround, since I spent some time scratching my head over this. I was able to get vtk display working with the OCCT IVtk data mapper class by creating a swig library:

```

class Visualization {
public:

    static vtkSmartPointer<vtkPolyDataMapper> getDataMapper(const TopoDS_Shape& sh) {
        IVtkOCC_Shape* vtkOccShape = new IVtkOCC_Shape(sh);

        vtkSmartPointer<IVtkTools_ShapeDataSource> occSource = vtkSmartPointer<IVtkTools_ShapeDataSource>::New();

        occSource->SetShape(vtkOccShape);

        vtkSmartPointer<vtkPolyDataMapper> mapper = vtkSmartPointer<vtkPolyDataMapper>::New();
        mapper->SetInputConnection(occSource->GetOutputPort());
        mapper->SetResolveCoincidentTopologyToPolygonOffset();
        return mapper;
    }
};

```

Then in your swig file:
```

%define VTK_SMARTPOINTER(vtk_t)
%{
#include <vtk_t##.h>
%}
%typemap(out) vtkSmartPointer<vtk_t>
{
    vtkSmartPointer<vtkObjectBase> castObj = static_cast<vtkSmartPointer<vtk_t>>($1);
    $result = vtkPythonUtil::GetObjectFromPointer(castObj);
}
%typemap(typecheck, precedence=SWIG_TYPECHECK_POINTER) vtk_t*
{
  $1 = vtkPythonUtil::GetPointerFromObject($input,#vtk_t) ? 1 : 0;
}
%enddef

VTK_SMARTPOINTER(vtkPolyDataMapper)
VTK_SMARTPOINTER(vtkOpenGLPolyDataMapper)

class Visualization {
public:

    static vtkSmartPointer<vtkPolyDataMapper> getDataMapper(TopoDS_Shape& sh);
};
```

Finally in python you can use it like so:

```
dm = vtk_occ_bridge_swig.Visualization_getDataMapper(shape)
actor = vtkmodules.vtkRenderingCore.vtkActor()
actor.SetMapper(dm)
```

(This is an adaptation of the vtk swig interface files found here https://github.com/burlen/vtk.i  )