tpaviot/pythonocc-core #653: Is is possible to pack pythonocc to exe file?

ecila92, 2019-05-11:
Hi guys!
I need help with packaging my pythonocc program containing pyQt5
I tried to make exe file with pyinstaller but the exe file does not works and this warning massage comes out when I try to start exe file.

 File "tesetui_2.py", line 27, in <module>
ImportError: No module named 'core_geometry_face_recogition_from_stepfile'
[22140] Failed to execute script tesetui_2

I think this is causing because there is no hook file for pythonocc and oce core.
I tried to build the hook file for pythonocc, but it was too hard for me as just a student...

Can anyone tell me how to make exe file from .py containing pythonocc and pyQt5...
There is no need to use pyinstaller, any program will do fine!
Thank you!

waterbug, 2019-05-11:
1)  Google (or duckduckgo, or other favorite search engine ...) is your 
friend.

2)  There are a lot of suggestions about pyinstaller and pythonocc 
buried in the pythonocc
issue tracker -- for example:

https://groups.google.com/forum/#!msg/pythonocc/dvdsLLw2REM/-u0_m8OeDgAJ

Some day they will hopefully be extracted into a document by some 
civic-minded
person (maybe me if I ever find time! ;).

Sorry I can't offer a more specific suggestion.  As to how you can get 
more specific
help:  in your message, specify what package manager you are using (pip 
or conda
and what version), and what versions of pyqt and pythonocc.

Steve

waterbug, 2019-05-11:
P.S.  This is an issue-tracker and what you posted is not really an "issue"
but a question, so you should be posting to the pythonocc google group:

pythonocc@googlegroups.com

Steve

ecila92, 2019-05-11:
Thank you for your kind suggestion and I'm sorry to post at wrong section T.T
please forgive me it was my first time posting something on github and am bit desperate!T.T

Durksz, 2019-05-14:
Your question is answered in #553 .

adam-urbanczyk, 2019-07-13:
I think it might be interesting for the community to look at a complete working example for a non-trival PyQt app using PythonOCC:

https://github.com/CadQuery/CQ-editor/blob/master/pyinstaller.spec

The trick was to bundle certain files and write a runtime hook that will define OCC specific variables:

https://github.com/CadQuery/CQ-editor/blob/master/pyinstaller/pyi_rth_occ.py

jf---, 2019-07-30:
curious if anyone has tried [pyoxidizer](https://github.com/indygreg/PyOxidizer) which seems to be the new kid in town

adam-urbanczyk, 2019-08-02:
Looks interesting indeed, bu no experience on my side.

Durksz, 2019-08-02:
I have no experience with pyoxidizer unfortunately

Durksz, 2019-08-07:
The faster startup-time can be a nice benefit! I have not have tried it though