tpaviot/pythonocc-core #654: ModuleNotFoundError: No module named '_BRepPrimAPI'

TristanDoherty, 2019-05-15:
Fresh Conda environment with pythonocc-core=0.18.2, jupyterlab=0.35.6, Windows 10 x64.

I'm trying to run the jupyter examples and this import statement fails:
`from OCC.Display.WebGl.jupyter_renderer import JupyterRenderer`

`
---------------------------------------------------------------------------
ImportError                               Traceback (most recent call last)
~\miniconda3\envs\occ\lib\site-packages\OCC\Core\Bnd.py in swig_import_helper()
     45         try:
---> 46             return importlib.import_module(mname)
     47         except ImportError:

~\miniconda3\envs\occ\lib\importlib\__init__.py in import_module(name, package)
    125             level += 1
--> 126     return _bootstrap._gcd_import(name[level:], package, level)
    127 

~\miniconda3\envs\occ\lib\importlib\_bootstrap.py in _gcd_import(name, package, level)

~\miniconda3\envs\occ\lib\importlib\_bootstrap.py in _find_and_load(name, import_)

~\miniconda3\envs\occ\lib\importlib\_bootstrap.py in _find_and_load_unlocked(name, import_)

~\miniconda3\envs\occ\lib\importlib\_bootstrap.py in _load_unlocked(spec)

~\miniconda3\envs\occ\lib\importlib\_bootstrap.py in module_from_spec(spec)

~\miniconda3\envs\occ\lib\importlib\_bootstrap_external.py in create_module(self, spec)

~\miniconda3\envs\occ\lib\importlib\_bootstrap.py in _call_with_frames_removed(f, *args, **kwds)

ImportError: DLL load failed: The specified procedure could not be found.

During handling of the above exception, another exception occurred:

ModuleNotFoundError                       Traceback (most recent call last)
<ipython-input-2-e90c5f884fa2> in <module>
----> 1 from OCC.Display.WebGl.jupyter_renderer import JupyterRenderer

~\miniconda3\envs\occ\lib\site-packages\OCC\Display\WebGl\jupyter_renderer.py in <module>
     43     sys.exit(0)
     44 
---> 45 from OCC.Core.Bnd import Bnd_Box
     46 from OCC.Core.BRepPrimAPI import BRepPrimAPI_MakeSphere
     47 from OCC.Core.BRepBndLib import brepbndlib_Add

~\miniconda3\envs\occ\lib\site-packages\OCC\Core\Bnd.py in <module>
     47         except ImportError:
     48             return importlib.import_module('_Bnd')
---> 49     _Bnd = swig_import_helper()
     50     del swig_import_helper
     51 elif _swig_python_version_info >= (2, 6, 0):

~\miniconda3\envs\occ\lib\site-packages\OCC\Core\Bnd.py in swig_import_helper()
     46             return importlib.import_module(mname)
     47         except ImportError:
---> 48             return importlib.import_module('_Bnd')
     49     _Bnd = swig_import_helper()
     50     del swig_import_helper

~\miniconda3\envs\occ\lib\importlib\__init__.py in import_module(name, package)
    124                 break
    125             level += 1
--> 126     return _bootstrap._gcd_import(name[level:], package, level)
    127 
    128 

ModuleNotFoundError: No module named '_Bnd'
`

TristanDoherty, 2019-05-28:
I'm not sure why this fixed things, but uninstalling and re-installing oce solved the issue.
conda remove oce
conda install -c tpaviot -c oce -c pythonocc pythonocc-core=0.18.2 python=3.6

closed 2019-05-28 by TristanDoherty

Song-xx, 2022-02-25:
> conda remove oce

oce? PackagesNotFoundError