tpaviot/pythonocc-core #658: How to Clear or Update the Display Screen

akverma26, 2019-06-03:
I am creating an interactive cube in which I can drag a vertex of the cube and change the shape.
What I am doing for this, on `MouseMoveEvent` I am taking 2D Coordinates of the cursor and then converting into 3D Space using `self._display.View.ConvertWithProj(cursor.x(),cursor.y())` and then I am changing coordinates of the clicked vertex with new calculated coordinates and draw a cube with this new vertex and other old 7 vertices but I am not able to delete the previous cube, Hence there are all previous cubes and there is a mesh.

Can anyone suggest me how to delete previous cubes or how to clear the entire display screen?

tpaviot, 2020-01-05:
Use the ```EraseAll``` method:
```self._display.EraseAll()```

closed 2020-01-05 by tpaviot

Tanneguydv, 2021-04-08:
Hello,
I have the exact same issue and the EraseAll method is not ok cause it erases all what's display.
For example if you want to keep another piece in your display it's not possible.

I've seen that @henrikrudstrom has created an EraseShape method in 2011 using dictonnary and list.
https://github.com/tpaviot/pythonocc/commit/ac560e94fc088cece0ea6df5b49b26a2bbd3eb55?branch=ac560e94fc088cece0ea6df5b49b26a2bbd3eb55&diff=unified

@tpaviot you responded to him "But why not create an AdvancedOCCViewer class, that inherits from OCCViewer, and provide additional and/or experimental features."
Is there a way to test simply this feature? It could be very useful for a lot of cases and I'm too new to PythonOCC to edit it as Henrik has done.

Thank you