tpaviot/pythonocc-core #660: STEP export - slow

Antoine333, 2019-06-17:
Hello,

i tried to export the shape of a stair (IfcStair - FZK Haus) to a step file by using the function 'write_step_file'  (src/Extend/DataExchange.py). It took about half an hour to write a file with 214423 ents.

[image]

Why does the export of this shape take that long? Are there any kind of shapes, or special parts of the geometry which slow down the export?

Antoine333, 2019-07-18:
@tpaviot Thanks for your reply!
The IfcStair is part of the loaded ifc model "FZK Haus" - 2.5 MB (http://www.ifcwiki.org/index.php?title=KIT_IFC_Examples).
I created the shape of the stair by using IfcOpenShell.
The exportet step file size is with 12.8 MB bigger then the full ifc model.

Antoine333, 2019-07-18:
The part of the geometry which slows down the export seems to be the railing.  It is a 'TopoDS_Shape' (Type:Compound) built from 2082 'TopoDS_Shape's (Type:Face) - not from solids.

The export of the other parts take just a second:
[image]

tpaviot, 2020-01-05:
You can try with the latest version (7.4.0rc1) and give your feedback here. Please also insert your python code to check whether the slow speed comes from your code or your machine.