tpaviot/pythonocc-core #662: OCC.Display.SimpleGui does not work for pythonocc 0.18.2

adurukan, 2019-07-03:
I was working with **python 2.7** and **pythonocc 0.18.1** before and the method:
`from OCC.Display.SimpleGui import init_display` was working fine. However, I had to change my system to **python 3.6** and **pythonocc 0.18.2** since **pythonocc 0.18.1** cannot be installed on my linux 64 bit system. The issue on that can be seen at #649 
Now, when I try `from OCC.Display.SimpleGui import init_display`, the following error appears:

```
ImportError                               Traceback (most recent call last)
~/miniconda3/envs/myenv/lib/python3.6/site-packages/OCC/Core/AIS.py in swig_import_helper()
    129         try:
--> 130             return importlib.import_module(mname)
    131         except ImportError:

~/miniconda3/envs/myenv/lib/python3.6/importlib/__init__.py in import_module(name, package)
    125             level += 1
--> 126     return _bootstrap._gcd_import(name[level:], package, level)
    127 

~/miniconda3/envs/myenv/lib/python3.6/importlib/_bootstrap.py in _gcd_import(name, package, level)

~/miniconda3/envs/myenv/lib/python3.6/importlib/_bootstrap.py in _find_and_load(name, import_)

~/miniconda3/envs/myenv/lib/python3.6/importlib/_bootstrap.py in _find_and_load_unlocked(name, import_)

~/miniconda3/envs/myenv/lib/python3.6/importlib/_bootstrap.py in _load_unlocked(spec)

~/miniconda3/envs/myenv/lib/python3.6/importlib/_bootstrap.py in module_from_spec(spec)

~/miniconda3/envs/myenv/lib/python3.6/importlib/_bootstrap_external.py in create_module(self, spec)

~/miniconda3/envs/myenv/lib/python3.6/importlib/_bootstrap.py in _call_with_frames_removed(f, *args, **kwds)

ImportError: libGL.so.1: cannot open shared object file: No such file or directory

During handling of the above exception, another exception occurred:

ModuleNotFoundError                       Traceback (most recent call last)
<ipython-input-11-4aaab5e56803> in <module>
----> 1 from OCC.Display.SimpleGui import init_display

~/miniconda3/envs/myenv/lib/python3.6/site-packages/OCC/Display/SimpleGui.py in <module>
     24 from OCC import VERSION
     25 from OCC.Display.backend import load_backend, get_qt_modules
---> 26 from OCC.Display.OCCViewer import OffscreenRenderer
     27 
     28 log = logging.getLogger(__name__)

~/miniconda3/envs/myenv/lib/python3.6/site-packages/OCC/Display/OCCViewer.py in <module>
     28 
     29 import OCC
---> 30 from OCC.Core.AIS import AIS_Shape, AIS_Shaded, AIS_TexturedShape, AIS_WireFrame
     31 from OCC.Core.TopoDS import TopoDS_Shape
     32 from OCC.Core.gp import gp_Dir, gp_Pnt, gp_Pnt2d, gp_Vec

~/miniconda3/envs/myenv/lib/python3.6/site-packages/OCC/Core/AIS.py in <module>
    131         except ImportError:
    132             return importlib.import_module('_AIS')
--> 133     _AIS = swig_import_helper()
    134     del swig_import_helper
    135 elif _swig_python_version_info >= (2, 6, 0):

~/miniconda3/envs/myenv/lib/python3.6/site-packages/OCC/Core/AIS.py in swig_import_helper()
    130             return importlib.import_module(mname)
    131         except ImportError:
--> 132             return importlib.import_module('_AIS')
    133     _AIS = swig_import_helper()
    134     del swig_import_helper

~/miniconda3/envs/myenv/lib/python3.6/importlib/__init__.py in import_module(name, package)
    124                 break
    125             level += 1
--> 126     return _bootstrap._gcd_import(name[level:], package, level)
    127 
    128 

ModuleNotFoundError: No module named '_AIS'
```

Any help would be appreciated.

sfornengo, 2019-07-15:
Same problem for me... Did you find a workaround ?

RRijnbeek, 2019-08-05:
Same problem.  It works for me with  Python 2.7. But It doesn't workwork for me with python 3.x versions.

I use WIN 10 64 bits

RRijnbeek, 2019-08-06:
I find a solution:

Here i put the sequence of the conda installation for an environment where ifcopenshell, and pythonocc 0.18.2 works without problems. 

conda create --name <NEW_ENV> python=3
activate <NEW_ENV>
conda install -c conda-forge -c oce -c dlr-sc -c ifcopenshell ifcopenshell
conda install -c tpaviot -c conda-forge -c dlr-sc -c pythonocc -c oce -c 3dhubs pythonocc-core==0.18.2 python=3.6
conda install pyqt
conda install wxpython

Now i can work and visualize  pointcloud , 3D models and ifcopenshell in the same environment on Windows 10 and python 3.6.7