tpaviot/pythonocc-core #667: Update pip package

roipoussiere, 2019-08-05:
The [pythonocc-core pip package](https://pypi.org/project/pythonocc-core/) has not been updated since Dec 4, 2014 and use pythonocc-core v0.16. It should be updated to the last version v0.18.1.

jf---, 2019-08-05:
we should remove this package indeed, or update it with a message suggesting to install via conda instead.

gggggbbbbb, 2019-08-06:
Please update instead of removing it ;)

gggggbbbbb, 2019-08-10:
@jf--- @tpaviot is there actually any plan to update the pip package (or provide access to .whl files)? It would be incredibly helpful for a project I am currently developing where I can't use Anaconda packages.

sim0nx, 2019-08-19:
would indeed be nice to update the pypi package instead of removing it :-)

Peque, 2019-09-24:
@jf--- @tpaviot Friendly ping. :innocent: 

Do not know how hard it would be, but would you accept a PR to fix this? Or is it simply that you do not want to support non-Conda installations at all? If so, is there a technical reason/limitation?

It seems to me like Python wheels could help nowadays distributing the package through PyPI, but I do not know much about them, so I may be missing something.

ANGEL1303, 2019-09-30:
Hello, does Python OCC work in python 3.7 or 3.8?

Why doesn't it have been updated?

traverseda, 2020-03-31:
I'm not a conda user, is there any benefit to making something a conda package instead of a pip package? I've honestly never seen that done before.

tpaviot, 2020-04-01:
Related to #796

tpaviot, 2020-04-01:
@traverseda a bit of history. When I started pythonocc, in 2008 (12 years ago !!), pip was only able to install 100% pure python packages. It was not able to install precompiled binary packages that contain c or c++ code. At that time, pip used to download the source code and launch the build process on your machine. It was perfect on Linux, where a c/c++ compiler is installed by default, but not for Windows users: Windows users usually don't have a c/c++ compiler installed. Most of the questions I had were : "How to I install mingw32 on my machine ?". A nightmare.

In 2010, when the project started to spread, the question was not "pip" or "conda" or whatever, but: "what's the best way to distribute precompiled binaries for each platform ?". conda was born to solve this issue. It saved many packages maintainers life. Later, python egg/wheels provided the same solution. It became possible to easily install pythonocc using conda, everybody was ok with that. Nobody asked for another solution, since this one does the job.

Now, here is the point: creating/maintaining a pip package is a dozens-of-hours commitment per year. I have serious doubts that this investment would bring any valuable improvement, compared to the same amount of time I could spend on other issues. Just convince me. Or take the lead on this topic.

Peque, 2020-04-01:
> Just convince me.

@tpaviot In an attempt to do so... :innocent: 

PyPI packages are more accessible, and I think that is a pretty good reason. If you have Conda, you can install packages with `pip`, but it is not possible the other way around. Some people may not have Conda.

You could argue that those people can install Conda, but sometimes that is not possible. Nowadays many people deploy projects to the cloud and many cloud services do not offer Conda-based images (at least not in their "free" tiers"). Some offer Docker-based images (which allows you to use Conda or whatever you want), but at a different price (usually higher of course). This can hurt even toy projects. I have been in a situation where I did not deploy a toy project simply because it was going to cost me money if deployed with the Conda requirement (and I was of course not going to make any money out of it).

You do have a maintenance burden when adopting pip, but maybe it is not much more than the burden you have for Conda (I may be wrong here though, since I have no real experience packaging wheels). What I would suggest is not to have both packages, but only the wheels, since that works for both `pip` and `conda` users. Hopefully the hours you spend packaging for `pip` are compensated by the hours you do not spend packaging for `conda`. :crossed_fingers:

rainman110, 2020-04-01:
The main argument against wheels is, that you can't bundle binary only packages.

Conda allows this and this is the reason why we started creating conda packages.

Also, conda provides a nice build infrastructure to create packages, which is not given otherwise (especially compiled packages).

One other issue is the maintenance overhead @tpaviot already mentioned. Building 1 pythonocc configuration already takes an hour. Adding another hour for occt. Multiply this with the number of python versions, architectures and operating systems. 

It could be possible to combine the occt and pythonocc conda packages into a wheel, avoiding recompilation. This is IMHO the only practical way for a small team of maintainers with limited compute resources and spare time.

Peque, 2020-04-01:
@rainman110 Thanks for the clarification. Being an OSS project, did you have a look at GitHub actions to maybe take care of that building process for you? Limits [seem to be generous](https://help.github.com/en/actions/getting-started-with-github-actions/about-github-actions#usage-limits).

rainman110, 2020-04-01:
We're already running on azure which is pretty similar to Github Actions.

tpaviot, 2020-04-04:
Interesting talk related to conda-press here https://www.youtube.com/watch?v=ovMqBLspkK4 @scopatz

Dogrtt, 2020-08-28:
Hi there. I tried to use `conda-press` for building `whl` from `conda` packages. Firstly, keep in mind [this](https://github.com/conda-incubator/conda-press/issues/50) issue about `Python38`. Also, if you use `--fatten`  option to glue packages up into single `whl`, `pip` will not be able to install this package because there is several `*.dist-info` in single `whl`. If build it without `--fatten`, you will get the separated `whl` s for each of sub-packages (`fontconfig`, `vc`, `freetype`, `tbb`, etc.) but in this case you will not be able just run `pip install pythonocc_core-7.4.0-0_*.whl`, because `pip` is not able to search in the current folder for different dependencies (of course), so you should done it manually in specific order. Then, after manual installing of all packages from `whl's`  you will get the errors about missed `DLL`'s such as `TKernel` and so on, that's because all of this libs was placed in the `venv/Lib/site-packages/Library/bin` and by default `venv` doesn't looks in this place. So what's your thoughts about it?
P.S. I wasn't able to run conda-press with installed CPython... by default, in active `conda env` when you call `conda press` it looks this script in your `CPython` folder... It's very annoying.

scopatz, 2020-08-28:
PRs to conda-press are very welcome!

CsatiZoltan, 2020-09-09:
Another reason for supporting pip: Read the Docs. When you host documentation on [Read the Docs](https://readthedocs.org/), you can choose your build system: either pip **xor** conda. Locally, on your system, you can of course use both. But not RTD.

adam-urbanczyk, 2020-09-09:
So what is the problem with conda and RTD that would be solved by pip?

CsatiZoltan, 2020-09-09:
@adam-urbanczyk More packages are available on PyPI. In my project, there are packages that can only be installed through pip.

adam-urbanczyk, 2020-09-09:
Well, you can use conda+pip (also on RTD), take a look e.g. here:

https://github.com/CadQuery/cadquery/blob/2d721d0ff8a195a0902eb9c3add88d07546c05b1/environment.yml#L23
https://github.com/CadQuery/cadquery/blob/master/.readthedocs.yaml

CsatiZoltan, 2020-09-09:
Thanks! I didn't know about the `pip:` directive in conda environment files.

traverseda, 2020-10-01:
So the main reason why you should support pip is so that I can add `pythonocc-core` to my python project's requirements. Where as if this project uses conda, well anything built on top of it is forced to use conda as well.

That's a pretty big difference, and opens up the potential for this to work with projects like python-appimage and pyinstaller.

CsatiZoltan, 2020-10-01:
What's more, the conda package manager is often extremely slow and unsuccessful in resolving dependencies, in which case it does not allow installing the other packages either.

traverseda, 2020-10-01:
It also doesn't seem to work with existing virtualenvs? Or at least I can't get it to work. So that means no pipenv, no poetry, no pipx, etc.

Using conda locks you out of a lot of a lot of the python ecosystem, and I think that's only going to get worse over time.

adam-urbanczyk, 2020-10-02:
I don't know about python-appimage, but you can definitely bundle packages with pyinstaller from a conda env. I used it to build a pythonocc based app with success.

If this is so important for you, you should consider contributing to e.g. conda-press (see above).

rainman110, 2020-10-02:
There are good reasons to support pip, without any doubt!

But there are currently technical limitations or at least difficulties that makes the creation of wheel packages very hard.

So instead of focusing on the pip vs conda discussion, we should more on how we actually make pip packages!

@traverseda Feel free to contribute!

tpaviot, 2020-10-02:
Each package management system is very specific, and require strong technical skills as well as a long experience. Nobody here has ever decided to drop pip support in favor of conda. Five years ago, it was easier to get done with conda, that's all. And then we had to maintain the package, and everybody lived happy untill @roipoussiere open this issue :blush:

There are plenty of cool package management systems: pip, apt, yum, brew, choco etc. Impossible to provide packages for all of them. I hope someday someone will try to standardize this mess, but in the meantime indeed anyone is invited to contribute a pip/anything else package.

traverseda, 2020-10-02:
Right, so I don't have a lot of experience with build systems, aside from occasionally changing a few flags here and there. My day job and my failing business keep me pretty busy these days, but it's great to know that's something the maintainers here *are* interested in.

Looking through the install directions I notice that it relies on particular libraries installed on the host OS? Most of the time with wheel packages I've seen them pull down *all* the sources and build them. Basically do a static build and not rely on the OS package manager at all. It's my understanding that that makes the whole thing a whole lot easier to work with, once you actually get it running.

Does that make sense? Having the install download/include all it's dependencies? Or is that going to end up being too much? Basically I'm imagining that if we can get cmake to build the whole thing as a completely static library, not really link against any host packages, that should be most of the battle.

Anyone following this issue with significant cmake experience? Because that's where I'm going to struggle the most.

Other than that, I'm not sure what technical limitations you're expecting, like I say I don't really have experience with conda or what problems it's solving. I think if we can get the cmake stuff working I should be able to get it building/packaging from the `setup.py`, and therefore working as both a source (for the more exotic architectures) and binary package.

But like I say, I don't *really* know what I'm talking about here, so please feel free to correct me if I'm oversimplifying or misunderstanding anything.

>There should be one-- and preferably only one --obvious way to do it.

I think pip is probably a good option, I don't think anyone is going to be concerned about you not providing apt or yum packages, that's a job for distro maintainers. I also don't imagine anyone would be too worried about you not providing conda packages if a pip package is available? The goal here would be to try and replace the conda package with a pip package? Maintaining both seems like it would be a pain.

adam-urbanczyk, 2020-10-03:
> I also don't imagine anyone would be too worried about you not providing conda packages if a pip package is available? The goal here would be to try and replace the conda package with a pip package? Maintaining both seems like it would be a pain.

Avoiding making monolithic packages like this is exactly why conda was invented. With conda you can pull OCCT from e.g. conda-forge and just use it as your dependency. I imagine some of the current users _will_ be worried, because such a monolithic pip package will possibly break their env.

traverseda, 2020-10-03:
What do you mean by monolithic? You saying that we'd need to also make gmesh a package, and link against that?

adam-urbanczyk, 2020-10-03:
PythonOCC obviously depends on OCCT, which in itself has nothing to do with Python at all (it is a C++ lib). I assumed that you want to distribute it in your package, hence the word monolithic. That could result in issues if I used another pip package that also distributs (possibly incompatible) OCCT.

traverseda, 2020-10-03:
>That could result in issues if I used another pip package that also distributes (possibly incompatible) OCCT.

I mean obviously it wouldn't install the compiled OCCT to the same place on the OS as the native package manager. It would install it into a folder belonging to that particular package, meaning you could have different versions of the library in different virtual-envs and all that stuff.

At no point should a pip package be installing statically-compiled files anywhere but into their own package namespace.

My understanding is that the conda package is already distributing it that way, and it's only the compile step that relies on the native package manager (apt/yum/whatever) having OCCT/SMESH installed? That's what I understood from going through `install.md`.

adam-urbanczyk, 2020-10-03:
I'm not referring to the OS but to a different python package depending on OCCT. On linux you can use rpath for isolation, but what do you intend to do on win? Static linking or adding UUIDs to the DLL names? I'm just trying to point out that it will likely become a mess rather fast.

Conda  BTW is not working like that - it has a separate package for OCCT and a separate packages for PythonOCC which depends on the former. The compile step (conda build) actually uses conda to fetch the deps, so you can rely on other people providing OCCT builds.

waterbug, 2020-10-03:
On 10/2/20 6:48 PM, traverseda wrote:

> <snip> .... I also don't imagine anyone would be too worried about you 
> not providing conda packages if a pip package is available? The goal 
> here would be to try and replace the conda package with a pip package? ..
>

I, for one, would be *very* unhappy to have a pip package instead of a 
conda package.
Conda is far superior to pip, especially in managing complex packages with
C, C++, and other libraries, which is the main motivation for developing 
conda in the
first place:  scientific and engineering packages often involve complex 
library
dependencies that pip is not good at managing.

If you don't know about that, you should read some of the history of
conda -- it would be worth your time, especially if you are going to 
continue
editorializing about it.

Steve

traverseda, 2020-10-03:
>If you don't know about that, you should read some of the history of conda -- it would be worth your time, especially if you are going to continue editorializing about it.

Can you explain a bit more about how not providing a conda package would impact your workflow? Does your environment not use any pip packages? As an end-user of the library, how does not packaging via conda actually change anything? Are you swapping out the shared libraries each package depends on? I get that for a long time it was pretty much the only way to distribute binary packages but other than that I'm having a hard time finding any advantages for end users, other than it being an alternative/wrapper for virtualenv's.

One of the big advantages of pip for binary packages is that you can have a binary distribution (wheel) and source distribution (sdist) in one pip package. So that means if I wanted to install a package on a weird CPU architecture, it will just build the package for that architecture. I've found that very useful when working on projects in the embedded world, particularly using buildroot-based linux distros with AArch64 numpy.

I mean before you're too dismissive about the pip ecosystem you should take a moment to understand *why* a lot of users want it, aside from just not wanting to be forced into the conda ecosystem.

>Static linking or adding UUIDs to the DLL names?

How does this conda package solve that problem? It does allow you to install different versions of the package into different environments right? Is it bundling it's own linker?

CMake is definitely an area I'm weak in, but I'd probably try to follow along with something like [this](https://martinopilia.com/posts/2018/09/15/building-python-extension.html). Most of these presume a single shared-library per python package. Not sure if that would be a deal breaker, as I'm still having trouble understanding what advantages the conda package provides, aside from being easier on the maintainers.

A lot of bigger packages do have a conda package *and* a pip package, I think it's a lot easier to go from wheel to conda than the other way around. I agree that for a package like this it would probably be a lot of work. Basically it would mean maintaining two completely different build systems, which is not something I think anyone would want.

waterbug, 2020-10-03:
> If you don't know about that, you should read some of the history
>     of conda -- it would be worth your time, especially if you are
>     going to continue editorializing about it.
>
> Can you explain a [...]
I do not use any pip packages, I exclusively use conda.  The conda 
package manager comes with its
own "pip", so you /can/ do "pip install" inside a conda environment.  
However, if you want to build
a conda package that has dependencies for which conda packages are not 
available, you will want
to create conda packages for them -- fortunately, if they are pure 
python packages that is almost
trivially easy:  conda has a command "skeleton" that can create a conda 
"recipe" from a PyPI
package, and then just run "conda build [package]" on the recipe, done.  
I have done that many
times, since I use only conda to build my packages.
> One of the big advantages of pip for binary packages is that you can
> have a binary distribution (wheel) and source distribution (sdist) in
> one pip package. So that means if I wanted to install a [...]
I don't find that to be an advantage, but then I do not work in the 
"embedded world", so I can't speak to that.
I only deploy my apps on Linux, Windows, and OSX.
>
> I mean before you're too dismissive about the pip ecosystem you should 
> take a moment to understand /why/ a lot of users want it, aside from 
> just not wanting to be forced into the conda ecosystem.
I understand a lot about the pip ecosystem, since I have been 
programming in Python for more than
20 years so I only used pip until a few years ago when conda was invented.
>
>     Static linking or adding UUIDs to the DLL names?
>
> How does this conda package solve that problem? It does allow you to 
> install different versions of the package into different environments 
> right? Is it bundling it's own linker?
It does bundle a lot but if you are interested in that level of detail 
you need to consult the docs.
Conda is quite powerful and can isolate its virtual environments (which 
are different from pip's)
extremely well.
> CMake is definitely an area I'm weak in, but I'd probably try to
> follow along with something like this
> <https://martinopilia.com/posts/2018/09/15/building-python-extension.html>.
> Most of these [...]
Actually I think it's easier to go from conda package to wheel -- 
earlier in this thread, Thomas
included a link to Anthony Scopaz's excellent presentation on "conda 
press", I strongly
recommend watching it if you are interested in this level of detail!
https://www.youtube.com/watch?v=ovMqBLspkK4

traverseda, 2020-10-03:
>I do not use any pip packages, I exclusively use conda. 

I mean that's just not how 99% of the python world operates.

Are there any end user features you'd be missing if this was a pip package? Are you like dynamically swapping out shared libraries? Are you linking directly to specific shared libraries from c/cpp programs embedded along side the python code?

traverseda, 2020-10-03:
I will give @scopatz talk a watch when I have a moment later tonight. I'm a big fan, "xonsh" is my primary shell.

adam-urbanczyk, 2020-10-03:
> How does this conda package solve that problem? It does allow you to install different versions of the package into different environments right? Is it bundling it's own linker?
>
> CMake is definitely [...]

I'm not referring to how conda is isolating envs, I'm asking you how do you want to isolate different packages (say NetGen and PythonOCC) that depend on OCCT (shared lib) installed in the **same** env. To my understanding there is no good way of doing this with pip, because it is not (does not want to be?) a "general" package manager. Conda, on the contrary, is and allows to just install an OCCT package.

YaroslavKormushyn, 2022-07-18:
> Hi there. I tried to use `conda-press` for building `whl` from `conda` packages. Firstly, keep in mind [this](https://github.com/conda-incubator/conda-press/issues/50) issue about `Python38`. Also, [...]

We use Azure Functions on Python, so we're limited to only `pip` in remote builds. Microsoft kind of [swept under the rug](https://github.com/Azure/azure-functions-core-tools/issues/1031) the possibility of a separate package manager.

I tried to `conda-press` the `pythonocc-core` package into a wheel, but got to the point of missing DLLs and left it at that.
Made a working linux-64 environment locally, used pip and conda to install all necessary dependencies, and made it work locally. Deployed to remote linux-64 environment with a `--no-build` option (function + all the dependencies as a single standalone package) only to find that it still doesn't work on the remote environment, because I cannot guess the exact agent architecture and configuration in a consumption plan (I build on Ubuntu 20.04 and the agent is of an unknown linux-64 arch).

How would one approach the task of building a standalone wheel for pythonocc-core for all of the existing linux-64 packages available on conda-forge? I'm at my wit's end here.

roipoussiere, 2022-09-12:
Just want to point out that `cadquery-ocp` is now published on PyPI: https://pypi.org/project/cadquery-ocp/

adam-urbanczyk, 2025-05-22:
> I get that back then conda was the best choice to distribute binary packages, but this is not the case anymore.

Honest question: why is it not the case anymore? Are the issues mentioned here https://pypackaging-native.github.io/ solved?

sanzoghenzo, 2025-05-23:
> > I get that back then conda was the best choice to distribute binary packages, but this is not the case anymore.
> 
> Honest question: why is it not the case anymore? Are the issues mentioned here https://pypackaging-native.github.io/ solved?

Sorry for my bad wording. Yes they are still issues.
But many projects mentioned have solved those and provide wheels with all the dependencies needed, at least for what I see (using pandas/geopandas and ifctoolkit daily on windows and wsl ubuntu).

tpaviot, 2026-10-19:
#1502