tpaviot/pythonocc-core #670: about 18.2 and 18.3 releases

roipoussiere, 2019-08-06:
From [the readme](https://github.com/tpaviot/pythonocc-core#oce-and-pythonocc):

> The basis of pythonocc is a python wrapper for the oce C++ library / CAD kernel, aka pythonocc-core. pythonocc-core version number matches the oce library releases its wrapping.
> 
> For example; the current pythonocc-core release, 0.18, requires any of the OCE 0.18.x releases. Here, the Major version name of either OCE or pythonocc-core release is 0, the Minor version is 18 and the Patch version is (optionally) x. pythonocc-core can be built with any OCE version that has a corresponding Major and Minor version number. For example, pythonocc-core 0.18.1 can be built with OCE-0.18.1.

Currently, the [latest oce release is 18.3](https://github.com/tpaviot/oce/releases) (released this on 14-02-2018), but the [latest pythonocc-core release](https://github.com/tpaviot/pythonocc-core/releases) is 18.1 (released this on 15-12-2017).

There are many commits published since (282, as of 07-08-2019), but I note that [there is no milestone](https://github.com/tpaviot/pythonocc-core/milestones?state=open) for an eventual 18.2 or 18.3 release.

So 2 questions:

- are you planning to publish a 18.2 and/or 18.3 release(s)?
- should I use the 18.1 pythonocc-core release from 15-12-2017 among with the 18.1 oce release from 11-05-2017, or use latest version of both repositories, ignoring the message in the readme?

dumblob, 2019-11-10:
Releasing newer versions is getting more and more important.

Any plans to release newer version? Or should we rather switch to the "more suitably" maintained fork at https://github.com/CadQuery/pythonocc-core ?

rainman110, 2019-11-10:
We are working on a 0.19 release that is based on OCCT 7.4.0. You can expect the release in the next couple of weeks.

jf---, 2019-11-11:
| We are working on a 0.19 release that is based on OCCT 7.4.0

Given that this release will address `OCCT` rather than `OCE`, can I suggest to have version parity, and make it explicit that this version addresses another source distribution? 

Given this chance, it might be worth to update the `README` of `OCE` to reflect what the project impetus was, many moons ago, and why the project perhaps becomes deprecated. I think making this explicit (maybe even on the OCCT forum ) helps in navigating the OS CAD/CAM ecosystem.

dumblob, 2019-11-27:
@rainman110 is all going well with the new release?

tpaviot, 2019-12-29:
Try the new release:

```
# first create an environment
conda create --name=whatever-name python=3.7
source activate whatever-name
conda install -c dlr-sc -c pythonocc pythonocc-core=7.4.0beta2
```

closed 2019-12-29 by tpaviot