tpaviot/pythonocc-core #676: Conda install error on windows 10

efirvida, 2019-09-09:
Hello I´m trying to install it on a conda env and I get this error:

```
C:\ conda install -c tpaviot pythonocc-core

Collecting package metadata (current_repodata.json): done
Solving environment: failed with initial frozen solve. Retrying with flexible solve.
Solving environment: failed with repodata from current_repodata.json, will retry with next repodata source.

Collecting package metadata (repodata.json): done
Solving environment: failed with initial frozen solve. Retrying with flexible solve.
Solving environment: /
Found conflicts! Looking for incompatible packages.
This can take several minutes.  Press CTRL-C to abort.
failed

UnsatisfiableError: The following specifications were found to be incompatible with each other:

Package sqlite conflicts for:
python=3.7 -> sqlite[version='>=3.25.3,<4.0a0|>=3.26.0,<4.0a0|>=3.27.2,<4.0a0|>=3.28.0,<4.0a0|>=3.29.0,<4.0a0']
Package oce conflicts for:
pythonocc-core -> oce==0.18.3
Package tbb conflicts for:
pythonocc-core -> tbb[version='>=2018_20171205']
Package openssl conflicts for:
python=3.7 -> openssl[version='>=1.1.1a,<1.1.2a|>=1.1.1b,<1.1.2a|>=1.1.1c,<1.1.2a']
Package pip conflicts for:
python=3.7 -> pip
Package smesh conflicts for:
pythonocc-core -> smesh==6.7.6
Package * conflicts for:
pythonocc-core -> *[track_features=vc14]
Package vc conflicts for:
python=3.7 -> vc[version='14.*|>=14.1,<15.0a0']
Package six conflicts for:
pythonocc-core -> six
Package freetype conflicts for:
pythonocc-core -> freetype[version='>=2.7,>=2.9.1,<3.0a0']
```

I'm using the latest conda package with python 3.7, any advice?

waterbug, 2019-09-09:
I have had similar problems with a python 3.7 conda env but have been
successful with a python 3.6 env -- try using a python 3.6 env:

     conda create -n [name] python=3.6

Steve

efirvida, 2019-09-11:
> I have had similar problems with a python 3.7 conda env but have been successful with a python 3.6 env -- try using a python 3.6 env:     conda create -n [name] python=3.6 Steve
> […](#)

Thanks for your answer but didn't work. ow I got this error:

```
Package smesh conflicts for:
pythonocc-core -> smesh==6.7.6
Package * conflicts for:
pythonocc-core -> *[track_features=vc14]
Package freetype conflicts for:
pythonocc-core -> freetype[version='>=2.7,>=2.9.1,<3.0a0']
Package pip conflicts for:
python=3.6 -> pip
Package vc conflicts for:
python=3.6 -> vc[version='14.*|>=14.1,<15.0a0']
Package oce conflicts for:
pythonocc-core -> oce==0.18.3
Package tbb conflicts for:
pythonocc-core -> tbb[version='>=2018_20171205']
Package sqlite conflicts for:
python=3.6 -> sqlite[version='>=3.25.3,<4.0a0|>=3.26.0,<4.0a0|>=3.29.0,<4.0a0']
Package six conflicts for:
pythonocc-core -> six
Package vs2015_runtime conflicts for:
python=3.6 -> vs2015_runtime
```

waterbug, 2019-09-11:
I could be wrong, but I'm not sure it is possible to install 0.18.3 on 
Windows.
I'm using pythonocc-core 0.18.2 (from the "tpaviot" channel).  I think 
that is what you will get
if you create a new, clean python 3.6 env and then "conda install 
pythonocc-core" -- I should
also note that I use a .condarc file with the following content to 
specify the channels:

channels:
   - tpaviot
   - oce
   - 3dhubs
   - dlr-sc
   - conda-forge
   - pythonocc
   - defaults

It seems to be important to put "conda-forge" channel before (i.e. 
higher priority) than "defaults".

efirvida, 2019-09-11:
Thanks a lot!! the **.condarc** file did the trick.
Best!

Omnipius, 2019-10-24:
For those having the same issue, the root cause is the new channel dependency 3dhubs, which is not listed in the typical pythonocc install instructions.

tpaviot, 2019-12-29:
Latest release should not require any additionnal channel:

```
conda create --name=whatever-name python=3.7
source activate whatever-name
conda install -c dlr-sc -c pythonocc pythonocc-core=7.4.0beta2
```

closed 2019-12-29 by tpaviot