tpaviot/pythonocc-core #680: init_display gives Segmentation fault (core dumped)

Vishnu-C, 2019-09-25:
installed pythonOCC for python 3.6 in Ubuntu 18.04.2 by

conda install -c conda-forge -c dlr-sc -c pythonocc -c oce pythonocc-core==0.18.1 python=3.6

running a simple demo with visualization gives Segmentation fault

On commenting the below line, I don't see the fault.
display, start_display, add_menu, add_function_to_menu = init_display()

Also tried with switching my Nvidia and Intel graphics card as active.

tpaviot, 2019-12-29:
Please update to the latest version, and reopen the issue if ever it still fails

```
# first create an environment
conda create --name=whatever-name python=3.7
source activate whatever-name
conda install -c dlr-sc -c pythonocc pythonocc-core=7.4.0beta2
```

closed 2019-12-29 by tpaviot

Sohaib90, 2020-03-16:
I was facing the same issue. I solved it by install wxPython which is responsible for the gui interface using anaconda in my pythonocc environment. I did not get the segfault after this. Hope this helps

tpaviot, 2020-03-16:
thanks @Sohaib90