tpaviot/pythonocc-core #687: Gear Meshing

efirvida, 2019-10-16:
Hello everyone:
Maybe this is not the right place to ask, for hints but is the place that I know ;)
I need to create an FEA model of spur-gears, let's say that I know how to model the gears, improving my project: https://github.com/efirvida/python-gearbox . then I want to mesh the gears like this pictures;

[image]

[image]

I know how to do all the cad procedures... but I don´t know If I can use PythonOCC to mesh the geometry as I want.

Thanks in advance

rainman110, 2019-11-08:
You probably should have a look at GMSH: http://gmsh.info/

It also supports geometries from OpenCASCADE.

tpaviot, 2019-12-29:
@efirvida Have a look at the ipython notebook about how to use both pythonocc and gmsh 
https://github.com/tpaviot/pythonocc-demos/blob/master/jupyter_notebooks/occ_triangle_mesh_gmsh.ipynb

Try online at
https://mybinder.org/v2/gh/tpaviot/pythonocc-binderhub/master and run jupyter_notebooks/occ_triangle_mesh_gmsh.ipynb

closed 2019-12-29 by tpaviot