tpaviot/pythonocc-core #688: Problems solving environment during Conda Install

DavidCadtec, 2019-10-17:
trying to install in my Linux (Python 3.5) using Conda Install ... and get:

"Initial quick solve with frozen env failed. Unfreezing env and trying again."
After sometime get
Solving environment: failed

and then after more time a huge list of Package setuptools conflicts...

Any way around ?
Thanks !

Omnipius, 2019-10-24:
There is an additional channel required now.

channels:
   - tpaviot
   - oce
   - 3dhubs
   - dlr-sc
   - conda-forge
   - pythonocc
   - defaults

Make sure you're calling all of those channels either in your install command or in condarc.

agnivade, 2019-10-25:
I am facing a similar issue too.

This is what I am doing

```
$conda install -c tpaviot -c oce -c 3dhubs -c dlr-sc -c conda-forge -c pythonocc -c defaults pythonocc-core==0.18.1 python=3.7
Collecting package metadata (current_repodata.json): done
Solving environment: failed with initial frozen solve. Retrying with flexible solve.
Solving environment: failed with repodata from current_repodata.json, will retry with next repodata source.
Collecting package metadata (repodata.json): done
Solving environment: failed with initial frozen solve. Retrying with flexible solve.
Solving environment: / 
Found conflicts! Looking for incompatible packages.
This can take several minutes.  Press CTRL-C to abort.
Examining freetype:  88%|████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████                 | 363/411 [03:06<02:45,  3.45s/- ]
Comparing specs that have this dependency:   0%|                                                                                                                                     | 0/23 [00:00<?, ?it/s]
Finding shortest conflict path for freetype:  58%|██████████████████████████████████████████████████████████████████████                                                  | 7/12 [1:54:24<55:12, 662.52s/it]
```

And the command is stuck at this for several hours now. 

Please advise.

DavidCadtec, 2019-10-27:
Tried:
conda install -c tpaviot -c oce -c 3dhubs -c dlr-sc -c conda-forge -c pythonocc -c defaults --force-reinstall pythonocc-core==0.18.1 python=3.5

And got those messages... (failure in the end ...)
Collecting package metadata (current_repodata.json): done
Solving environment: failed with current_repodata.json, will retry with next repodata source.
Initial quick solve with frozen env failed.  Unfreezing env and trying again.
Solving environment: failed with current_repodata.json, will retry with next repodata source.
Collecting package metadata (repodata.json): done

I'm  installing in a virtualenv (not a Condais  env...) is this somehow related to the issue

tpaviot, 2019-12-29:
@DavidCadtec Please use the latest version:

```
# first create an environment
conda create --name=whatever-name python=3.7
source activate whatever-name
conda install -c dlr-sc -c pythonocc pythonocc-core=7.4.0beta2
```

closed 2019-12-29 by tpaviot