tpaviot/pythonocc-core #690: How to build in windows in python=3.7

tnakaicode, 2019-10-18:
I tried to build and install pythonocc-core in python=3.7 in win10.

1. Build and install OCE (https://github.com/tpaviot/oce) by cmake.
2. OCE instlled "C:\OCE-0.18\"
3. Tried to build pythonocc-core (https://github.com/tpaviot/pythonocc-core) by cmake

But, I have following error.
```
CMakeFiles/_StdSelect.dir/objects.a(StdSelectPYTHON_wrap.cxx.obj):StdSelectPYTHON_wrap.cxx:(.text+0x3): undefined reference to `__imp__Py_NoneStruct'
```
I think the cause of this error is that cmake can't find OCE library.

Is there any good solution?

rainman110, 2019-11-01:
I think it is a problem linking the python libraries. Linking correctly the python libraries is a bit tricky.
Your output ``undefined reference to `__imp__Py_NoneStruct`` makes this pretty clear.

What value has your variable PYTHON_LIBRARY? It should point to the python library of your python interpreter.

@tpaviot We should use cmake new findpython modules, which solves this issue. See here:
https://gitlab.kitware.com/cmake/cmake/issues/17664#note_643362

@tpaviot  We should now link python as follows:
```cmake
find_package(Python3 COMPONENTS Development)

swig_add_library(mymodule LANGUAGE python SOURCES mymdule.i)
target_link_libraries(mymodule Python3::Module)
```

Small downside: CMake's FindPython requires CMake 3.12. I still recommend to update the cmake scripts, as this potentially solves multiple problems.

tpaviot, 2019-11-01:
@rainman110 That means officially drop support for python2, I'm ok with that

rainman110, 2019-11-01:
Yes! Python 2 will only be supported until end of the year. It's time to move on :smile:

tnakaicode, 2019-11-08:
I set windows variables following, and solved this problem.

```
PYTHON_LIBRARY=C:\Users\Nakai\Anaconda3\libs\python37.lib;C:\Windows\System32;
```

And, I add following lines in set Installing directroy in CMakefiles.txt. 
If this addition is not available, OCC will be installed in C:/OCC. I want to install to "~/Anaconda3/Lib/site-packages/"

```
execute_process(COMMAND ${PYTHON_EXECUTABLE} -c "from distutils.sysconfig import get_python_lib; import os;print(get_python_lib())" OUTPUT_VARIABLE python_lib OUTPUT_STRIP_TRAILING_WHITESPACE )
string(REPLACE "\\" "/" python_lib "${python_lib}")
set(PYTHONOCC_INSTALL_DIRECTORY ${python_lib}/OCC)
```

But, I have another problem.