tpaviot/pythonocc-core #697: Make gyroid surface with pythonOCC

bragostin, 2019-11-05:
I am trying to create a gyroid surface with pythonOCC (like in the first attached image), using ```GeomAPI_PointsToBSplineSurface()``` like in the following code. But what I get is, shown in the second image, does not look correct. Is it possible at all to create a triply periodic surface with GeomAPI_PointsToBSplineSurface, or have I taken a completely wrong direction?

```python
import math
from OCC.Core.gp import gp_Pnt
from OCC.Core.BRepBuilderAPI import BRepBuilderAPI_MakeEdge, BRepBuilderAPI_MakeFace, BRepBuilderAPI_MakeWire
from OCC.Core.TColgp import TColgp_Array2OfPnt
from OCC.Core.GeomAPI import GeomAPI_PointsToBSplineSurface
from OCC.Core.GeomAbs import GeomAbs_C2
from OCC.GeomAbs import GeomAbs_Intersection
from OCC.BRepOffset import BRepOffset_MakeOffset,BRepOffset_Skin
from OCC.Display.SimpleGui import init_display
display, start_display, add_menu, add_function_to_menu = init_display()
import numpy as np
from numpy import sin, cos, pi

def gyroid(x, y, z, t):
    return cos(x)*sin(y) + cos(y)*sin(z) + cos(z)*sin(x) - t

lattice_param = 1.0
strut_param = 0.0
resolution = 11j
res = int(np.imag(resolution))
x, y, z = pi/2. * np.mgrid[-1:1:resolution, -1:1:resolution, -1:1:resolution] * lattice_param
vol = gyroid(x, y, z, strut_param)
xv, yv, zv = vol[0].flatten(), vol[1].flatten(), vol[2].flatten()
gxyz  = [(x,y,z) for x,y,z in zip(xv, yv, zv)]

x = np.array([range(res)])
y = np.array([range(res)])
kx, ky = np.meshgrid(x, y)
kxy  = [[kx,ky] for kx,ky in zip(kx.flatten(), ky.flatten())]

def face():
    
    array = TColgp_Array2OfPnt(1, res, 1, res)
    
    for i in range(len(gxyz)):
        x, y, z = gxyz[i]
        kx, ky = kxy[i]
        pt = gp_Pnt()
        pt.SetCoord(float(x), float(y), float(z))  
        array.SetValue(int(kx)+1, int(ky)+1, pt)
        
    curve = GeomAPI_PointsToBSplineSurface(array, 3, 8, GeomAbs_C2, 0.001).Surface()
    
    red_face = BRepBuilderAPI_MakeFace(curve, 1e-6).Face()

    display.DisplayShape(red_face, update=False)
    
if __name__ == '__main__':
    face()
    start_display()
```
[image]
[image]

tnakaicode, 2019-11-05:
How about this code?

```
import numpy as np
import math

from OCC.Display.SimpleGui import init_display
from OCC.gp import gp_Pnt
from OCC.BRepBuilderAPI import BRepBuilderAPI_MakeEdge, BRepBuilderAPI_MakeFace, BRepBuilderAPI_MakeWire
from OCC.TColgp import TColgp_Array2OfPnt
from OCC.GeomAPI import GeomAPI_PointsToBSplineSurface
from OCC.GeomAbs import GeomAbs_C2
from OCC.GeomAbs import GeomAbs_Intersection
from OCC.BRepOffset import BRepOffset_MakeOffset, BRepOffset_Skin

def gyroid(x, y, z, t):
    return np.cos(x)*np.sin(y) + np.cos(y)*np.sin(z) + np.cos(z)*np.sin(x) - t

if __name__ == '__main__':
    display, start_display, add_menu, add_function_to_menu = init_display()

    lat = 1.0
    res = 11
    pt = 0.0

    x, y, z = np.pi/2. * np.mgrid[-1:1:res*1j, -1:1:res*1j, -1:1:res*1j] * lat
    vol = gyroid(x, y, z, pt)

    print(vol.shape[-1])
    for iz, vz in enumerate(vol[::, ]):
        print(iz)
        ixy = vz.shape
        pts = TColgp_Array2OfPnt(1, ixy[0], 1, ixy[1])
        for idx, val in np.ndenumerate(vz):
            ix, iy = idx[0], idx[1]
            px, py, pz = x[ix, iy, iz], y[ix, iy, iz], val
            pnt = gp_Pnt(px, py, pz)
            #display.DisplayShape(pnt)
            pts.SetValue(ix+1, iy+1, pnt)

        crve = GeomAPI_PointsToBSplineSurface(
            pts, 3, 8, GeomAbs_C2, 0.001).Surface()
        face = BRepBuilderAPI_MakeFace(crve, 1e-6).Face()
        display.DisplayShape(face)
    
    display.FitAll()
    start_display()

```

[image]

bragostin, 2019-11-05:
It already looks better, although there are 11 surfaces instead of 1. Also the surface should have 6 curved edges as shown in the matplotlib figure below, not 4. I would think that the x and y from ```vol```  rather than ```mgrid``` should be used though.

[image]

rainman110, 2019-11-08:
What you try to do is not possible, at least not exactly. All CAD surfaces that are directly supported are parametric surfaces with 2 variables, e.g something like s(u,v).

Your gyroid function looks very different, like an implicit function. If the shape of the gyroid is really that complicated as in your upper image, you should probably decompose the patch into quadrilateral sub-patched and approximate each of the sub-patches.

I don't think that ```GeomFill_ConstrainedFilling``` will help you here with good results. What also may work is the Geom_Plate algorithm. It will return a trimmed surface, such that it is possible to have more than 4 surrounding edges of the surface.

Still, this type of surface is not very CAD friendly due to reasons mentioned above.

bragostin, 2019-11-09:
@rainman110  thanks for the tips! Yes, to have an exact representation of this function is not possible, but an approximated surface that looks like it should be possible because I have seen some examples made with FreeCad. 
Indeed ```GeomPlate_BuildPlateSurface``` looks promising, I am going to try with this function.
The example file core_geometry_surface_from_curves.py looks also interesting to achieve this.

bragostin, 2019-11-15:
@rainman110  I have followed your advice and found a solution, thank you for your help! Using ```scipy.measure.marching_cubes_lewine``` to decompose the surface into sub-patches  in conjunction with the OCC functions ```BRepBuilderAPI_MakePolygon``` to define the surface edges and ```BRepFill_Filling``` to create the surface produces the picture below. The code is in the gyroid_py-OCC.py.txt in case somebody finds it useful. 

It's not perfect yet, the construction of the surface 6 edges is a bit of craft work and would probably not work with other triply periodic surfaces. Also any resolution beyond 6 points will crash the code. I suppose ```BRepFill_Filling``` is then not able to solve all the constrains. But it does the job good enough!

[gyroid_py-OCC.py.txt](https://github.com/tpaviot/pythonocc-core/files/3850586/gyroid_py-OCC.py.txt)

[image]

closed 2019-11-19 by bragostin