tpaviot/pythonocc-core #699: OCCt 7.4.0: Adapt Viewer to changed API

rainman110, 2019-11-08:
For example, in OccViewer.py the following line

https://github.com/tpaviot/pythonocc-core/blob/adcbd7cfcd1bb810afeec61a7dc8ce0d80776ea5/src/Display/OCCViewer.py#L494

must be changed to

```Python
shape_to_display.SetMaterial(Graphic3d_MaterialAspect(material))
```

There are probably other occurences as well.

rainman110, 2019-11-11:
@tpaviot I am missing two commit on branch review/occt7x:

 - https://github.com/tpaviot/pythonocc-core/commit/d0ee410f3ca8dc3ce49f75171c06d3f2219091d1
 - https://github.com/tpaviot/pythonocc-core/commit/4759d25ddbddd5fad8b5dba866489e780e9f98dc

did you force push?

tpaviot, 2019-11-12:
>    - d0ee410
>    - 4759d25

Yes, probably. Did you commit directly to review/occt7x or on your own
branch ?

If you commit on review/occt7x, of course I will stop force pushing.

rainman110, 2019-11-12:
I committed directly into your repository of the current branch.

tpaviot, 2019-11-12:
ah ok, sorry, I did not expect you would push directly to the pythonocc-core repository. I guess that's why I had to git push force yestarday, my local copy was not up-to-date. Can you please re-push your two commits please ? Next time I will be more careful

commit 94b18bd9 (2019-11-18): Fixed remaining issues on OccViewer

Fixes #703
Fixes #699
closed 2019-11-18 by rainman110