tpaviot/pythonocc-core #702: Error in openning STEP a d IGES

mohrezarezaei, 2019-11-12:
Hi
I want to use python occ to read a STEP file and change to shape into voxel representation but in the first step, in reading the file, i keep getting the "Standard Out Of Range" error, I also tried to read an IGES file and got the same problem. It is also worth considering that i  have to skip the "Transfer Root" command as if i run it, my python stops operation. I use python 3.6 in windows 10 in Pycharm IDE and use Rhino 6 as my cad software. 
I hopefully look forward to your help

rainman110, 2019-11-12:
Can you please post a code snippet or provide the step file (if this is possible)?

Without anything, it's very hard for us to reproduce / understand, what kind of error you have.

mohrezarezaei, 2019-11-12:
sorry for being late. here is the code:

from OCC.STEPControl import STEPControl_Reader
step_reader = STEPControl_Reader()
step_reader.ReadFile("d:\a1.stp")
#step_reader.TransferRoot()
bl = step_reader.Shape()

the error is :

Traceback (most recent call last):
  File "C:\Users\sarir\Anaconda3\lib\site-packages\IPython\core\interactiveshell.py", line 2862, in run_code
    exec(code_obj, self.user_global_ns, self.user_ns)
  File "<ipython-input-2-d2a47f1ada6c>", line 1, in <module>
    runfile('C:/Users/sarir/.PyCharmCE2019.1/config/scratches/scratch_1.py', wdir='C:/Users/sarir/.PyCharmCE2019.1/config/scratches')
  File "C:\Program Files\JetBrains\PyCharm Community Edition 2019.1.3\helpers\pydev\_pydev_bundle\pydev_umd.py", line 197, in runfile
    pydev_imports.execfile(filename, global_vars, local_vars)  # execute the script
  File "C:\Program Files\JetBrains\PyCharm Community Edition 2019.1.3\helpers\pydev\_pydev_imps\_pydev_execfile.py", line 18, in execfile
    exec(compile(contents+"\n", file, 'exec'), glob, loc)
  File "C:/Users/sarir/.PyCharmCE2019.1/config/scratches/scratch_1.py", line 5, in <module>
    bl = step_reader.Shape()
RuntimeError: Standard_OutOfRange

and my STEP File is attached
[ap1.zip](https://github.com/tpaviot/pythonocc-core/files/3838254/ap1.zip)