tpaviot/pythonocc-core #703: pythonocc7.4.0 beta Quantitiy Color bug.

liuxin2322, 2019-11-14:
```python
from OCC.Core.BRepPrimAPI import BRepPrimAPI_MakeBox
from OCC.Display.OCCViewer import rgb_color
#box
my_box = BRepPrimAPI_MakeBox (10,10,10).Shape()
if __name__ == "__main__":
    from OCC.Display.SimpleGui import init_display
    display, start_display, add_menu, add_function_to_menu = init_display()
    display.DisplayShape(my_box,update=True,color=rgb_color(0,1,0))
    start_display()
```
the code works well in pythonocc0.18.2, but bug happen in latest version.
```
TypeError: in method 'new_Quantity_Color', argument 1 of type 'Quantity_NameOfColor'
Additional information:
Wrong number or type of arguments for overloaded function 'new_Quantity_Color'.
  Possible C/C++ prototypes are:
    Quantity_Color::Quantity_Color()
    Quantity_Color::Quantity_Color(Quantity_NameOfColor const)
    Quantity_Color::Quantity_Color(Standard_Real const,Standard_Real const,Standard_Real const,Quantity_TypeOfColor const)
    Quantity_Color::Quantity_Color(NCollection_Vec3< float > const &)
```
install command:`conda create -n pythonocct  -c conda-forge -c tpaviot -c 3dhubs -c pythonocc -c dlr-sc -c oce pythonocc-core=7.4.0beta python=3.7`
os: ubuntu16.06 linux64

liuxin2322, 2019-11-14:
oops,other example also occur bugs

liuxin2322, 2019-11-14:
@liuxin2322 read_step_file_names_colors()

rainman110, 2019-11-14:
Actually, this Beta version was not meant to be installed yet. I am working on adapting the demo examples to the new API of opencascade 7.4.0.

liuxin2322, 2019-11-14:
@rainman110 ok, i'm looking forward to seeing it:grimacing:

commit 94b18bd9 (2019-11-18): Fixed remaining issues on OccViewer

Fixes #703
Fixes #699
closed 2019-11-18 by rainman110