tpaviot/pythonocc-core #704: conda build for pythonocc-7.4.0

tpaviot, 2019-11-14:


rainman110, 2019-11-14:
Here's the already working template:
https://github.com/DLR-SC/tigl-conda/tree/master/python-occ-7x

tpaviot, 2019-11-14:
@rainman110 your recent fix actually finds Python better than previous version (I've been monitoring travis build). I did not try your recipe, but I saw you managed to build packages for all oses. Congrat's, that's a tough work. you can open a PR and submit your recipe as a whole if you want.

tpaviot, 2019-11-14:
I should hide swig warnings in the conda recipe, there are tons of output polluting the console

rainman110, 2019-11-14:
Keep fingers crossed that we can now finally build windows.

In my recipe, I have warnings disabled.

Will make a PR!

rainman110, 2019-11-14:
Wow... Windows builds now out of the box on my machine! 😄 

Next milestone achieved!!!

rainman110, 2019-11-14:
I'll trigger a new CI build.

rainman110, 2019-11-15:
@tpaviot I have seen that you are now playing with conda and travis and the opengl settings.

Implementing this correctly can be very difficult imho. The problem is, that if you build your conda package againt Ubuntu's libgl* it might not work under other linux distros. Typical errors are undefined references or version mismatches.

To workaround this, pythonocc should either link against conda gl packages or against CDT packages of conda that provide a compatibility layer around the system libraries:
You can read more about this here: https://docs.conda.io/projects/conda-build/en/latest/resources/define-metadata.html#host

Long story short: Don't install systems opengl dev libaries, use condas.

tpaviot, 2019-11-15:
Are you sure "playing" is the right word ? :blush: I agree with you. I think 4im gonna play in another branch, there might be several dumb commits.

tpaviot, 2019-11-20:
Fixed, see issue #711

closed 2019-11-20 by tpaviot