tpaviot/pythonocc-core #708: Initializing OffscreenRenderer results in a segmentation fault

amarschn, 2019-11-18:
Hello,

I am getting a strange failure when trying to run the following on Ubuntu 18.04:
```
from OCC.Display.OCCViewer import OffscreenRenderer
o = OffscreenRenderer()
```
When trying to run this, I get the following output:

```
 ###### 3D rendering pipe initialisation #####
Display3d class initialization starting ...
Segmentation fault (core dumped)
```

I am not sure what exactly is going on, I have tried this with both Python2 and Python3 installations of PythonOCC (installed using `conda install -c conda-forge -c dlr-sc -c tpaviot -c oce -c pythonocc pythonocc-core=0.18.2`). 

If anyone has any ideas on things to check, it would be much appreciated as I am stumped at this point.

amarschn, 2019-11-18:
I've been looking into this more and I believe the issue has something to do with the graphics drivers set up on my system. Currently I am using a VPS for all testing out PythonOCC, and my guess is that something about how OpenGL or the graphics driver is configured is interfering with PythonOCC's ability to render.

amarschn, 2019-11-20:
Update:

amarschn, 2019-11-20:
Accidentally closed the issue, but it is not fixed just yet.

tpaviot, 2019-11-20:
We've been updating pythonocc to the next release. Can you please test again with a beta package:

```conda install -c dlr-sc/label/dev pythonocc-core ```

and report the current issue status ?

amarschn, 2019-11-21:
Hi Thomas,

I tried to install the beta package but I run into some dependency issues. I tried installing `python-cc core` in new Conda environments (I tried Python 3.5,3.6, and 3.7), but ended up with the same error:

```
UnsatisfiableError: The following specifications were found to be incompatible with each other:

Package sqlite conflicts for:
python=3.7 -> sqlite[version='>=3.24.0,<4.0a0|>=3.25.2,<4.0a0|>=3.25.3,<4.0a0|>=3.26.0,<4.0a0|>=3.27.2,<4.0a0|>=3.29.0,<4.0a0|>=3.30.1,<4.0a0']
Package pip conflicts for:
python=3.7 -> pip
Package libgcc-ng conflicts for:
python=3.7 -> libgcc-ng[version='>=7.2.0|>=7.3.0']
pythonocc-core -> libgcc-ng[version='>=7.3.0']
Package ncurses conflicts for:
python=3.7 -> ncurses[version='>=6.1,<7.0a0']
Package libffi conflicts for:
python=3.7 -> libffi[version='>=3.2.1,<4.0a0']
Package zlib conflicts for:
python=3.7 -> zlib[version='>=1.2.11,<1.3.0a0']
Package pyqt conflicts for:
pythonocc-core -> pyqt
Package libglu conflicts for:
pythonocc-core -> libglu
Package libstdcxx-ng conflicts for:
pythonocc-core -> libstdcxx-ng[version='>=7.3.0']
python=3.7 -> libstdcxx-ng[version='>=7.2.0|>=7.3.0']
Package openssl conflicts for:
python=3.7 -> openssl[version='>=1.0.2o,<1.0.3a|>=1.0.2p,<1.0.3a|>=1.1.1a,<1.1.2a|>=1.1.1b,<1.1.2a|>=1.1.1c,<1.1.2a|>=1.1.1d,<1.1.2a']
Package readline conflicts for:
python=3.7 -> readline[version='>=7.0,<8.0a0']
Package opencascade conflicts for:
pythonocc-core -> opencascade==7.4.0
Package tk conflicts for:
python=3.7 -> tk[version='>=8.6.7,<8.7.0a0|>=8.6.8,<8.7.0a0']
Package xz conflicts for:
python=3.7 -> xz[version='>=5.2.4,<6.0a0']
```

I have the following packages installed:
```
# Name                    Version                   Build  Channel
_libgcc_mutex             0.1                        main  
ca-certificates           2019.10.16                    0  
certifi                   2019.9.11                py37_0  
dbus                      1.13.12              h746ee38_0  
expat                     2.2.6                he6710b0_0  
fontconfig                2.13.0               h9420a91_0  
freetype                  2.9.1                h8a8886c_1  
glib                      2.63.1               h5a9c865_0  
gst-plugins-base          1.14.0               hbbd80ab_1  
gstreamer                 1.14.0               hb453b48_1  
icu                       58.2                 h9c2bf20_1  
jpeg                      9b                   h024ee3a_2  
libedit                   3.1.20181209         hc058e9b_0  
libffi                    3.2.1                hd88cf55_4  
libgcc-ng                 9.1.0                hdf63c60_0  
libpng                    1.6.37               hbc83047_0  
libstdcxx-ng              9.1.0                hdf63c60_0  
libuuid                   1.0.3                h1bed415_2  
libxcb                    1.13                 h1bed415_1  
libxml2                   2.9.9                hea5a465_1  
ncurses                   6.1                  he6710b0_1  
openssl                   1.1.1d               h7b6447c_3  
pcre                      8.43                 he6710b0_0  
pip                       19.3.1                   py37_0  
pyqt                      5.9.2            py37h05f1152_2  
python                    3.7.5                h0371630_0  
qt                        5.9.7                h5867ecd_1  
readline                  7.0                  h7b6447c_5  
setuptools                41.6.0                   py37_0  
sip                       4.19.8           py37hf484d3e_0  
sqlite                    3.30.1               h7b6447c_0  
tk                        8.6.8                hbc83047_0  
wheel                     0.33.6                   py37_0  
xz                        5.2.4                h14c3975_4  
zlib                      1.2.11               h7b6447c_3  
```

I still think that the OffscreenRenderer problem is likely related to an issue with my graphics setup on the server, due to the fact that `glxinfo` results in an error. Additionally, I've used the OffscreenRenderer in a separate Windows machine before without issue. I'll keep trying to figure it out though and I'll update this thread if I find a solution. I am also looking into whether I could do this in a Docker container, I found a thread in the Google Group that references this: https://groups.google.com/forum/#!topic/pythonocc/jjsVkDnOcCs, but it isn't clear that it was ever solved.

tpaviot, 2019-11-21:
@amarschn thank you for this feedback. The OffScreenRenderer requires an opengl driver, of course. The graphic pipe is just streamed to a file rather than a GUI. I dont understand exactly what you expect from a docker container, compared to your own machine, can you please elaborate ?

@rainman110 what's your opinion about the conda issue with the pythonocc-core7.4.0 package ?

amarschn, 2019-11-21:
@tpaviot I don't expect anything different from the docker container, I just thought I would mention it as that is how I would like to deploy the application I am working on that would use the offscreen renderer.

tpaviot, 2019-11-21:
@amarschn ok, I got it. The rendering must go through the opengl pipeline. The Offscreenrenderer is just something that automates a screen capture. If the server cannot render the geometry because it misses an opengl driver, you have to delegate this task to the client. It can be done using a webbrowser for instance.

amarschn, 2019-11-22:
I am now able to use the offscreen renderer! Looks like installing Xvfb did the trick. For anyone in the future who runs into this problem, these were the steps I followed, tested on DigitalOcean ubuntu droplet:

- Create DigitalOcean droplet
- Install conda (https://www.digitalocean.com/community/tutorials/how-to-install-anaconda-on-ubuntu-18-04-quickstart)
- Create conda environment: conda create --name occ python=3.7
- Activate conda environment: conda activate occ
- Install pythonocc-core
- Install mesa-utils graphics library
- Install ubuntu desktop
- Install xvfb
- Install pythonocc
- Set the display variable
- Configure xvfb
- Use `glxinfo` to test whether the virtual display is working

Most of the commands I ran when performing these installation/configuration steps:
```
$: sudo apt-get update
$: sudo apt-get install mesa-utils
$: sudo apt-get install ubuntu-desktop
$: sudo apt-get install xvfb
$: export DISPLAY=:99
$: Xvfb :99 -screen 0 640x480x24 &
$: glxinfo
```
Testing pythonocc:
```
>> from OCC.Display.OCCViewer import OffscreenRenderer
>> o = OffscreenRenderer()
```

closed 2019-11-22 by amarschn