tpaviot/pythonocc-core #709: OCCT 7.4.0: Undefined references when loading modules

rainman110, 2019-11-18:
import RWGltf
Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "/localdata1/sigg_ma/src/pythonocc-core/buildocct740/OCC/Core/RWGltf.py", line 18, in <module>
    from . import _RWGltf
ImportError: /localdata1/sigg_ma/src/pythonocc-core/buildocct740/OCC/Core/_RWGltf.so: undefined symbol: _ZTI29RWGltf_GltfLatePrimitiveArray

import RWMesh
Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "/localdata1/sigg_ma/src/pythonocc-core/buildocct740/OCC/Core/RWMesh.py", line 18, in <module>
    from . import _RWMesh
ImportError: /localdata1/sigg_ma/src/pythonocc-core/buildocct740/OCC/Core/_RWMesh.so: undefined symbol: _ZTI16RWMesh_CafReader

import RWObj
Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "/localdata1/sigg_ma/src/pythonocc-core/buildocct740/OCC/Core/RWObj.py", line 18, in <module>
    from . import _RWObj
ImportError: /localdata1/sigg_ma/src/pythonocc-core/buildocct740/OCC/Core/_RWObj.so: undefined symbol: _ZTI12RWObj_Reader

Additionally, I get an undefined reference from our (DLR-SC) conda builds:
import BRepBlend
Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "/localdata1/sigg_ma/miniconda/envs/pythonoccnew/lib/python3.7/site-packages/OCC/Core/BRepBlend.py", line 18, in <module>
    from . import _BRepBlend
ImportError: /localdata1/sigg_ma/miniconda/envs/pythonoccnew/lib/python3.7/site-packages/OCC/Core/_BRepBlend.so: undefined symbol: _ZNK20BRepBlend_AppSurface10TolReachedERdS0_

I don't know, why I don't get this error locally but just on the conda build, but it seems to be safe to me tu simply remove these functions from the wrapper.

rainman110, 2019-11-18:
It seems, that we have to remove RWObj module. I can't get rid of the undefined ref otherwise.

rainman110, 2019-11-18:
I still have the BRepBlend issue. This is related to the OCCT bug:
https://git.dev.opencascade.org/gitweb/?p=occt.git;a=commitdiff;h=746cb7c32872b2aa2a261f65b42cef3e826b1d6e

I assume, we have to also remove AppBlend::TolReached, as this is its abstract base class.

rainman110, 2019-11-18:
Alternatively, we could simply patch OCCT.

tpaviot, 2019-11-18:
Patching occt at oce would make sense. There are also many headers to patxh

rainman110, 2019-11-18:
On the other hand, that would make it very hard to use vanilla occt for the pythonocc users.

tpaviot, 2019-11-19:
I won't be hard, under the condition that patches does not introduce any API change nor binary incompatiblity

rainman110, 2019-11-19:
I applied the patch to the opencascade conda package at dlr-sc. This way, we could at least fix this issue for the conda users, which are probable the majority.

tpaviot, 2019-11-19:
I understand. I did the same with oce

closed 2019-11-19 by rainman110