tpaviot/pythonocc-core #710: Access primitive values that are returned by reference

rainman110, 2019-11-18:
Many functions like
```
double& math_Matrix::Value(i,j)
```

allows a read and write access its value by reference. Swig however simply returns an opaque pointer whose value cannot be accessed. This makes it impossible to read out e.g. Matrix values.

I see two options, to make this value accessible:
 1) Return by value and add a setter method. e.g.

   ```
   double math_Matrix::Value(i,j)
   void math_Matrix::SetValue(i,j, double)
   ```
   
 Drawback: required modification of the generator. could be tricky.

 2) Add generic setter and getter functions for ```double*```. This way we could read and write the value as follows:

```python
occ_set(matrix.Value(i,j), 5.)
print(occ_get(matrix.Value(i,j))) # 5.
```

 In which module would we add these accessor functions? How do we want to name them?

tpaviot, 2019-11-18:
2 is quicker to implement, but 1 is IMO better. occ_set would expose the c++ idioms at some point,  for instance when replying to the question 'in which case should I use occ_set'. First solution is simpler to use and to document : 'if the method exists, just use it'.

tpaviot, 2019-11-19:
@rainman110 FYI the first solution you suggest is already implemented in the generator, see https://github.com/tpaviot/pythonocc-generator/blob/wip/occt7x-wrapper/src/generate_wrapper.py#L1342
but it's limited to methods that don't take any parameter. I have to extend this to the case where the function takes any number of parameters, it should not be that hard.

tpaviot, 2019-11-19:
ok, the fix is ready, waiting for the build to complete

rainman110, 2019-11-19:
It seems, that the Getter and Setter are not created for math_Vector::Value. At least, i can't find them in the interface file.

tpaviot, 2019-11-20:
from ```math_Vector.hxx```:

```
//! accesses the value of index "theNum" of a vector.
  const Standard_Real& Value (const Standard_Integer theNum) const
  {
    return Array(theNum);
  }
```
In that case, only a getter should be wrapped, so setter is not defined. I did not take that into account.

tpaviot, 2019-11-20:
You're right, I don't know why the Value method is not wrapped. But it's not exactly the same case (getter+setter), it's only a getter that should be wrapped.

rainman110, 2019-11-20:
From what I see on the reference documentation, it has two value functions:
const Standard_Real& Value(idx) const;
Standard_Real& Value(idx);

Maybe, just the first is detected which would then even return a Float by value.

tpaviot, 2019-11-20:
Yes, that's it. The setter uses the () operator, which is excluded by default from the wrapper. It's something quite different from the gp_Mat issue.

tpaviot, 2019-11-20:
Yes, that's it. There's also a setter that uses the () operator, which is excluded by default from the wrapper. It's something quite different from the gp_Mat issue.

closed 2019-11-28 by tpaviot