tpaviot/pythonocc-core #711: Conda packaging. Fix conda-build for Linux/travis

rainman110, 2019-11-19:
@tpaviot Can you please explain, what the current issues are and what is required?

tpaviot, 2019-11-20:
I modified the conda yaml file as you suggest, i.e.

```
requirements:
  build:
    - {{ compiler('cxx') }}
    - {{ cdt('libx11-devel') }}         # [linux]
    - {{ cdt('xorg-x11-proto-devel') }} # [linux]
    - {{ cdt('mesa-libgl-devel') }}     # [linux]
```
but, on travis, cmake can't find opengl.

rainman110, 2019-11-20:
Does pythonocc use opengl directly? If not, we should remove this and also find_package(OpenGL).

rainman110, 2019-11-20:
By the way, the magic line to be able to find opengl is this here:
https://github.com/DLR-SC/tigl-conda/blob/16e9f21a630c10541fd0d6915301a8f571b83aea/python-occ-7x/build.sh#L5

rainman110, 2019-11-20:
More infos here:
https://github.com/ContinuumIO/anaconda-issues/issues/8779

tpaviot, 2019-11-20:
I tried to remove opengl from cmakelist, but some AIS headers include glx, I think it is required

tpaviot, 2019-11-20:
Fixed.

closed 2019-11-20 by tpaviot