tpaviot/pythonocc-core #713: bug:step part name return blank

liuxin2322, 2019-11-20:
version:pythonocc740beta  os:linux 64 
source code:
```python
from OCC.Display.SimpleGui import init_display
from OCC.Extend.DataExchange import read_step_file_with_names_colors
display, start_display, add_menu, add_function_to_menu = init_display()
shapes_labels_colors=read_step_file_with_names_colors('UiRS.1 - Robot Arm 3DOF.STEP')

for shpt_lbl_color in shapes_labels_colors:
    label, c = shapes_labels_colors[shpt_lbl_color]

    display.DisplayShape(shpt_lbl_color,color=c)
start_display()
```
console log :

```Name : 
Name : 
    shape color Name & RGB:  class<'Quantity_Color'> 307 0.7921568751335144 0.8196078538894653 0.9333333373069763
....
```

it seems part name return nothing or blank
test model:

`'UiRS.1 - Robot Arm 3DOF.STEP'`
[testStep.zip](https://github.com/tpaviot/pythonocc-core/files/3869053/testStep.zip)

hope  to solve this bug :)

tpaviot, 2019-11-20:
Honestly, I don't kow. I spent *dozains of hours*, during last summer, trying to figure out what's going on with CAF/XDE/TDF_Labels and so on, it's a pure mystery. I don't undesrtand how that work, definitely, and I'm afraid I can't help

liuxin2322, 2019-11-20:
@tpaviot :sob:   I want to help，but it's a huge project（maybe from begin to learn wrapper cpp）

tpaviot, 2019-12-13:
I tested with the STEP file provided, here is what I get.

[image]

Is it he expected colors ?

liuxin2322, 2019-12-13:
> I tested with the STEP file provided, here is what I get.
> 
> [image]
> 
> Is it he expected colors ?

It's weird .     using other web view  application, i get following pic:
[image]

tpaviot, 2019-12-13:
The lightning in simple display may affect shape rendering, I'll check with the x3d exporter

tpaviot, 2019-12-13:
Here is the result in a Chrome window

[image]

I'm not sure your renderer can be trusted more than mine. STEP and colors is a big deal. I used to work with a guy from NIST, he's working on a set of test files. Each software I tested (commercial Cat, SW, Autodesk inventor and free such as FreeCad), all of them render colors a different way.

liuxin2322, 2019-12-13:
@tpaviot you're right. different render display different，the returned data of color is mostly accurate because of x3d result.      but I have to say, qtdisplay may  have some implicit bugs

tpaviot, 2019-12-29:
Fixed.

closed 2019-12-29 by tpaviot

liuxin2322, 2019-12-31:
@tpaviot :+1: