tpaviot/pythonocc-core #714: Travis timeout issues

rainman110, 2019-11-20:
Try setting the following environment variable:

```bash
export PYTHONUNBUFFERED=1
```

I can't explain otherwise, why the pythonocc build on travis works on the dlr-sc repo but not here.

tpaviot, 2019-11-20:
Sometime travis builds freeze, but it's quite unpredictable. If I re-run some failing job one or two days after, then the build goes on.

The timeout issue is because pythonocc is too long to build on travis machines. That's why I'd like to speed up the build (precompiled headers, caching directories, use ccache etc.).

travis/appveyor are very convenient tools, especially because unittests are run. I'm  bit frustrated that I have to make the build on my own, it's a waste of time.

rainman110, 2019-11-20:
My point was, that I have no issues at all with travis and appveyor, even with very long runtimes:
 - https://ci.appveyor.com/project/rainman110/tigl-conda/builds/28945604
 - https://travis-ci.com/DLR-SC/tigl-conda/builds/137065144

rainman110, 2019-11-20:
@tpaviot The current build looks already much better. The job is getting killed after 70 minutes, not before.

I asked the Travis support to extend the build time per job to 120 minutes for the DLR-SC repo. This is usually no problem.

tpaviot, 2019-11-22:
@rainman110 setting PYTHONBUFFERED results in a better build, thank you. I asked for a 20mn extra time to travis support a couple of days ago, no answer yet.  I managed to get the appveyor build finally work, it appears it's much faster than travis (at least for windows). I wonder if it's the same for Linux. And I was notified, yesterday, that osx builds are now available https://www.appveyor.com/docs/macos-images-software/

In the coming days, I'll try to fix appveyor to support linux and osx. There's no need, IMO, to have both travis and appveyor since both support osx/windows/linux. If ever this attempt fails, I suggest we release a pythonocc-core 7.4.0beta anyway, we will use your repository for conda packages

rainman110, 2019-11-22:
I agree having both appveyor and travis is overkill. I personally used travis always for linux and macOS, appveyor for windows. 

The downside of appveyor is, that it does not offer parallel build for free. So this could again increase the total compilation time.

BTW, I played around with ccache and ccache-swig locally. I could make a clean build of pythonocc in around 10 seconds 😄

tpaviot, 2019-11-22:
I could easily add Ubuntu linux support for appveyor, with just a small change to the .yml configuration file. The build couldn't complete, compilation speed appears to be almost the same than travis.

10s to compile the whole project ? impressive !

tpaviot, 2019-11-22:
see https://ci.appveyor.com/project/tpaviot/pythonocc-core/build/job/ana459ga3wrd0r28

adam-urbanczyk, 2019-11-30:
@tpaviot  @rainman110  you could also consider azure - they have a  360 minute time limit per job. Take a look here for an example configuration: https://github.com/CadQuery/conda-packages which I use to build a fork of pythonocc.

tpaviot, 2019-11-30:
@adam-urbanczyk I registered an account on Azure just one week ago, I'm glad you already succeeded using it, it will save me a lot of time, thank you

tpaviot, 2019-12-03:
@adam-urbanczyk I successfully copied/pasted/modified your azure configuration, and was able to build pythonocc-core. Thank you, it's really impressive compared to appveyor/travis. How did you set up the anaconda token ? I did not find how to do it, in order to automate anaconda uploads.

@rainman110 you should have a look, see https://github.com/tpaviot/pythonocc-conda-azure

rainman110, 2019-12-03:
I've already seen it. Very impressive.

Is this free to use for open source projects?

adam-urbanczyk, 2019-12-03:
I defined the anaconda token as a secure env variable it in the webui  (
edit -> variables ).

The service is indeed free for os projects.

tpaviot, 2019-12-05:
Issue solved : Travis is not used anymore !

The whole ci process moved to MS Azure see https://dev.azure.com/tpaviot/pythonocc-core/_build?definitionId=2

closed 2019-12-05 by tpaviot