tpaviot/pythonocc-core #715: Build for 0.18.1 (18.04 Beaver) defaults to Python2, Need 3

jwatson-CO-edu, 2019-11-21:
I followed the below steps for installing OCE/Python-OCC, and to my dismay I found that the build process targets Python2 instead of Python3.  My lab's standard is Python 3.6+.  The instructions do not seem to address this issue.

1. Install OCE
`git clone git://github.com/tpaviot/oce.git
cd oce/
mkdir build && cd $_

cmake -DOCE_INSTALL_PREFIX:PATH=$HOME/libs/OCE  \
    -DOCE_WITH_FREEIMAGE:BOOL=ON \
    -DOCE_WITH_GL2PS:BOOL=ON \
    -DOCE_DRAW:BOOL=ON \
    -DOCE_WITH_FREEIMAGE:BOOL=ON \
    -DOCE_WITH_GL2PS:BOOL=ON \
    -DOCE_ADD_HEADERS:BOOL=ON \
    -DOCE_COPY_HEADERS_BUILD:BOOL=ON \
    ..

make -j8 
make install`

2. Install Python-OCC
`git clone git://github.com/tpaviot/pythonocc-core.git
cd pythonocc-core
mkdir cmake-build && cd $_

cmake -DOCE_INCLUDE_PATH=$HOME/libs/OCE/include/oce \
    -DOCE_LIB_PATH=$HOME/libs/OCE/lib \
    ..

make -j8
sudo make install`

rainman110, 2019-11-22:
Just open cmake-gui, activate advanced and grouped, and chose a different python interpreter and python library. 

Then click configure and then generate to change the makefiles.

Close cmake gui and build pythonocc as usual with
```
make -j 8
```

jwatson-CO-edu, 2019-11-24:
Oh, the instructions did not call for `cmake-gui`, but I did some learning and got the following results:
1. I was able to point `cmake-gui` at the correct installation of Python. However ...
1. I attempted to set vars `OCE_INCLUDE_PATH` and `OCE_LIB_PATH` as instructed.  Although their values appeared to those of my successful Py2.7 build, I received `make` errors indicating an incorrect OCE path.
1. Using the GUI, I was able to back out what `PYTHON_EXECUTABLE`, `PYTHON_INCLUDE_DIR`, and `PYTHON_LIBRARY` are supposed to be.  

**tl;dr**
I used the following `cmake` command: 
```
cmake -DOCE_INCLUDE_PATH=$HOME/libs/OCE/include/oce \
    -DOCE_LIB_PATH=$HOME/libs/OCE/lib \
    -DPYTHON_EXECUTABLE=/usr/bin/python3.6 \
    -DPYTHON_INCLUDE_DIR=/usr/include/python3.6 \
    -DPYTHON_LIBRARY=/usr/lib/x86_64-linux-gnu/libpython3.6m.so \
    ..
```
And was able to successfully import from OCC in Py3.

Overall the build process for this project is very straightforward, except for this snag.  I think if you provided some direction for Py3 users, it would be perfect.