tpaviot/pythonocc-core #719: Completeness w.r.t. OCCT 7.4 API

adam-urbanczyk, 2019-11-30:
Thanks for the hard work on wrapping 7.4! I did manage to build the latest master branch locally but I cannot import some classes. e.g. `ShapeUpgrade_UnifySameDomain`

https://www.opencascade.com/doc/occt-7.4.0/refman/html/class_shape_upgrade___unify_same_domain.html#ace3469ee91140ff11c17c56a365d7165

Is this a bug / feature / WIP ?

tpaviot, 2019-11-30:
I noticed this class is missing while trying to run the jupyter demos. I don't know yet if it was removed from upstream (an api change) or if it comes from the python wrapper

tpaviot, 2019-11-30:
You mention 'classes' with an ending 's'. Can you please report other missing classes?

adam-urbanczyk, 2019-11-30:
It seems to be in the API according to the docs (cf. link above).

tpaviot, 2019-11-30:
So related code has to be ported to the new api.

adam-urbanczyk, 2019-11-30:
Sorry @tpaviot , wrong wording: so far I'm stuck on `ShapeUpgrade_UnifySameDomain` . So it is one confirmed missing class for now.

tpaviot, 2019-11-30:
According to the html red, I see no reason why this class is not covered by the wrapper. I'm on my mobile phone right now, I'll be back on my laptop on monday to solve the issue. An announce the 740beta release at the same time

closed 2019-12-01 by tpaviot

adam-urbanczyk, 2019-12-02:
More on the same topic: I cannot find the `Font` module.

tpaviot, 2019-12-02:
I don't think the Font module has ever been wrapped

adam-urbanczyk, 2019-12-02:
Ok, then I guess it is better to continue the discussion here #723

commit b5870616 (2019-12-01): Add missing class ShapeUpgrade_UnifySameDomain, fix #719