tpaviot/pythonocc-core #720: handle can't be generated

silentmissile, 2019-12-01:
in previous 0.18.x version

a lot of classes has method: GetHandle() to generate the handle of the class

but now in version 7.4.0beta, all of them are lost

but many functions needs handle as the input parameter, so I can't continue my program

please tell me how to make handle of an object or fix this bug

rainman110, 2019-12-01:
Just mit GetHandle / GetObject. We changed everything under the hood, that you don't need this anymore. 

@tpaviot we should change the docstrings, to not confuse the users with the handle stuff.

tpaviot, 2019-12-02:
@silentmissile You have to port your code to the "transparent handles" feature. See for instance how we did with the bottle example at https://github.com/tpaviot/pythonocc-demos/commit/e59acdce5720d84ce76134789b48c268e36446d6#diff-68b70730ce65eb74e098809766ab3d0d

tpaviot, 2019-12-02:
@rainman110 docstrings don't refer to GetHandle or GetObject anymore, what would you change to the documentation ?

rainman110, 2019-12-02:
If a Handle is required for some function in C++, the docstring includes as an argument type Handle_XXX.

This could be misleading or at least confusing for the users.

tpaviot, 2019-12-02:
it's not Handle_XXX is the docstring, it's opencascade::handle<XXX>. I agree it has to be changed to XXX only

tpaviot, 2019-12-02:
@rainman110 What do you think of the following change template in all the docstrings:

First change : remove all 'const' or '&' information, which are not the matter of python users.
```
-       :type aShape: TopoDS_Shape &
+       :type aShape: TopoDS_Shape
```

Second change: replace opencascade handles with the base class name
```
-       :type aPres: opencascade::handle<Prs3d_Presentation> &
+       :type aPres: Prs3d_Presentation
        :param aDrawer:
-       :type aDrawer: opencascade::handle<Prs3d_Drawer> &
+       :type aDrawer: Prs3d_Drawer
        :param aVertex:
```

rainman110, 2019-12-02:
This would be perfect!

closed 2019-12-03 by tpaviot