tpaviot/pythonocc-core #722: TopTools_HSequenceOfShape gives error "raise AttributeError("No constructor defined")"

chenkianwee, 2019-12-02:
I have just updated to the 7.4.0beta version and I am receiving this error  "raise AttributeError("No constructor defined") when I use the TopTools_HSequenceOfShape . I can't really figure out what is wrong. I looked at the OCC documentation and it seems that the class requires TopTools_SequenceOfShape as an argument for version 7.4.0. I have included that but it still gives me this error. Here is a snippet of the code. I realise in 7.4.0 the Handle_TopTools_HSequenceOfShape no longer exist. So how do I obtain the handle from TopTools_HSequenceOfShape.  Any advise will be great thanks. 
```
#I first initial the TopTools_SequenceOfShape() 
toptool_seq_shape = TopTools_SequenceOfShape()
for edge in occedge_list: 
     toptool_seq_shape.Append(edge) 

#and use it as an argument for TopTools_HSequenceOfShape
edges = TopTools_HSequenceOfShape(toptool_seq_shape)

```

tpaviot, 2019-12-02:
can you please post a sample I could easily and quickly run to test ?

chenkianwee, 2019-12-02:
Here is a sample, thanks for the quick reply
`from OCC.Core.TopTools import TopTools_HSequenceOfShape, Handle_TopTools_HSequenceOfShape_Create, TopTools_SequenceOfShape
from OCC.Core.gp import gp_Pnt
from OCC.Core.BRepBuilderAPI import BRepBuilderAPI_MakeEdge

edge1_pts = [gp_Pnt(50.0, 0.0, 0.0), gp_Pnt(0.0, 50.0, 5.0)]
edge1 = BRepBuilderAPI_MakeEdge(edge1_pts[0], edge1_pts[1]).Edge()
edge2_pts = [gp_Pnt(0.0, 50.0, 5.0), gp_Pnt(0.0, 50.0, 0.0)]
edge2 = BRepBuilderAPI_MakeEdge(edge2_pts[0], edge2_pts[1]).Edge()
edge3_pts = [gp_Pnt(50.0, 0.0, 0.0), gp_Pnt(0.0, 50.0, 0.0)]
edge3 = BRepBuilderAPI_MakeEdge(edge3_pts[0], edge3_pts[1]).Edge()

edge_list = [edge1, edge2,edge3]

toptool_seq_shape = TopTools_SequenceOfShape()
for edge in edge_list: 
    toptool_seq_shape.Append(edge)  

edges = TopTools_HSequenceOfShape(toptool_seq_shape)
edges_handle = Handle_TopTools_HSequenceOfShape_Create() #? is this how I can get the handle?`

rainman110, 2019-12-02:
The short answer is, that you don't need the handle anymore. Just omit the last line and use edge instead of edge_handle

chenkianwee, 2019-12-02:
@rainman110 Thanks for the advise. The current issue I have is the line TopTools_HSequenceOfShape(toptool_seq_shape) gives me an error "raise AttributeError("No constructor defined")"  THanks

rainman110, 2019-12-02:
@tpaviot The generated code is missing a public keyword:

https://github.com/tpaviot/pythonocc-core/blob/534135b8feae9a9deb24eb9369d3838db4be1579/src/SWIG_files/wrapper/TopTools.i#L736

Therefore, the required constructor is not wrapped.

tpaviot, 2019-12-03:
Good catch !

closed 2019-12-03 by tpaviot

Moult, 2019-12-09:
Just ran into this today :) What triggers the build-bot to get a new build on https://anaconda.org/dlr-sc/pythonocc-core

tpaviot, 2019-12-09:
@Moult I just tagged a new release (7.4.0beta2). Latest python packages are available at:
https://anaconda.org/tpaviot/pythonocc-core

ping @rainman110

rainman110, 2019-12-09:
@Moult @tpaviot packages are built!

rainman110, 2019-12-09:
https://anaconda.org/DLR-SC/pythonocc-core/files?version=7.4.0beta2

commit 89e4147d (2019-12-03): Fixed wrong HSEQUENCE wrapper, fix #722