tpaviot/pythonocc-core #723: text to brep function no longer exist in the 7.4.0beta update?

chenkianwee, 2019-12-02:
Hi I am curios if the text to brep function is still around for the new update? If it does can someone point me to where is it located, I cant locate the it as the addons module does not exist in the 7.4.0beta update. Thanks.

tpaviot, 2019-12-03:
The text to brep code function was a piece of c++ code, wrapped in python in the Addons package. This was removed from the 740beta release. Maintaining c++ code inside pythonocc is a mess (think about the old Addons package, deprecated smesh support, visualization code etc.). In pythonocc, everything should be pure python. Please open a PR if ever you have something to contribute.

chenkianwee, 2019-12-03:
Thanks for the reply and explaining the situation.

tpaviot, 2019-12-03:
I reopen the issue, I will close it when it's possible to do the same with Python, that is to say when the Font module is available

Durksz, 2019-12-16:
I'm facing the same problem, which stops me from switchting to 7.4.0. 
Is there a way to create text in the 3d enviroment using opencascade?

tpaviot, 2019-12-16:
There are two solutions:

1. wrap the Font module. I tried to, but I can't wrap Font_BRepFont because of c++ issues. I'll try again, I did not realize how important it can be for some users (thanks for your feedback).

2. go the c++ way, as it used to be in previous releases.

If I don't succeed in 1., I'll go step 2. But, in 7.4.0, be sure you can finally use 3d BRep texts.

rainman110, 2019-12-16:
@Durksz Do you want 3D text or simply text in the 3D scene?

Otherwise, you can simply use the DisplayMessage method of the viewer, see e.g.
https://github.com/tpaviot/pythonocc-demos/blob/3ab791306e4569e8d4649b4ca3f074ea9141cff0/examples/core_geometry_quaternion.py#L58

Durksz, 2019-12-16:
I want to be able to export a solid with explaining text to step, such that when importing it into a cad program (freecad of NX for example) the specific parameters of the part are clear when importing.
This prevents issues when a tiny alteration is made of the solid which is not easily checked visually.

tpaviot, 2019-12-17:
I uploaded code for ```Font``` support to the branch review/wrap-font. Unfortunately there's an issue with the ```Font_BRepFont``` class (the most important one actually). I spent one hour looking at the .cxx code generated by swig and can't figure out the problem. So far.

tpaviot, 2019-12-24:
@Durksz I finally decided to port the Addon package to current pythonocc 7.4.0. See #739

Durksz, 2019-12-24:
Thanks!

I'll try it soon

closed 2019-12-27 by tpaviot

Durksz, 2019-12-29:
Can you update the PythonOCC on conda, such that I can test is easily?

rainman110, 2019-12-29:
I'd suggest to make a release candidate 1. Then I'll build the conda packages.

tpaviot, 2019-12-29:
I'm ok with that, I'll tag rc1tomorrow.

tpaviot, 2019-12-30:
@rainman110 Done, I tagged 7.4.0rc1

rainman110, 2019-12-30:
Conda builds are on the way

Durksz, 2019-12-30:
The text generation works!