tpaviot/pythonocc-core #729: Issues with AIS_InteractiveContext SetSelectionMode()

e3dEF, 2019-12-06:
Hello,
I'm using the OCCViewer.Viewer3d to display shapes and get feedback from the user by clicking them.
For example, I can successfully change the color of a shape after it was selected in the viewer.
```
clr = Quantity_Color.Name(1.0, 0, 0)
display.Context.SetColor(ais, clr)
```
Now I wanted to allow to select e.g. only edges of a shape. So I tried
`display.Context.SetSelectionMode(ais, TopAbs_EDGE)`
But it has no effect. I don't want to use,
`display.SetSelectionModeShape()`
which works, but is apllied on all shapes and additionally closes the context, so that I loose all links the shapes.
How can I change only selection mode of a single shape?

tpaviot, 2019-12-07:
@e3dEF I don't know.

rainman110, 2019-12-07:
The local context is not supported anymore. Please refer to the occt documentation for more information.

VASIMRLJJ, 2021-07-13:
I had the same problem, and that's how I solved it: simply add "self.Context.Deactivate()" before activate the context, refer to: [https://dev.opencascade.org/content/local-context-removal](https://dev.opencascade.org/content/local-context-removal)
`
    def SetSelectionMode(self, mode=None):
        self.Context.Deactivate()
        topo_level = next(modes)
        if mode is None:
            self.Context.Activate(AIS_Shape_SelectionMode(topo_level), True)
        else:
            self.Context.Activate(AIS_Shape_SelectionMode(mode), True)
        self.Context.UpdateSelected(True)

`

EmJay276, 2022-08-19:
Thanks @VASIMRLJJ I was looking for this fix for some time
Created a pull request https://github.com/tpaviot/pythonocc-core/pull/1134