tpaviot/pythonocc-core #733: Graphic3D_Material without texture not working in "core_display_textured_shape" example  example

Durksz, 2019-12-16:
I found another problem with the 7.4.0 version, adding material to th

When i remove the texture from the shape in the example:
"core_display_textured_shape" 

Thus change the following line 
from:
> display.DisplayShape(s.Shape(), material=m, texture=t)

to
> display.DisplayShape(s.Shape(), material=m)

I get the following error:

>   File "C:\Miniconda3\envs\pythonocc_test_env\lib\site-packages\OCC\Core\Graphic3d.py", line 3612, in __init__
>     _Graphic3d.Graphic3d_MaterialAspect_swiginit(self, _Graphic3d.new_Graphic3d_MaterialAspect(*args))
> 
> TypeError: in method 'new_Graphic3d_MaterialAspect', argument 1 of type 'Graphic3d_NameOfMaterial'
> Additional information:
> Wrong number or type of arguments for overloaded function 'new_Graphic3d_MaterialAspect'.
>   Possible C/C++ prototypes are:
>     Graphic3d_MaterialAspect::Graphic3d_MaterialAspect()
>     Graphic3d_MaterialAspect::Graphic3d_MaterialAspect(Graphic3d_NameOfMaterial const)

This error does not occur in the 0.18.2 version of PythonOCC.

tpaviot, 2019-12-28:
Thank you for the feedback, it's a part of code that has not been ported up to occt740

tpaviot, 2019-12-29:
@Durksz To apply material over a shape, just pass the material name as an argument, for instance:

```
display.DisplayShape(s, m=Graphic3d_NOM_SILVER)
```

materials and texture should not be set together. I updated demos, see https://github.com/tpaviot/pythonocc-demos/blob/master/examples/core_display_material_shape.py and https://github.com/tpaviot/pythonocc-demos/blob/master/examples/core_display_textured_shape.py

closed 2019-12-29 by tpaviot

Durksz, 2019-12-30:
Thanks for the updated demo's, Why I found it not working was because i want to be able to change the diffusion of the part, thus make it more of less reflective. Somewhere inbetween the material types Graphic3d_NOM_PLASTIC and Graphic3d_NOM_SHINY_PLASTIC. 

In the previous version i did this by setting the following:
m = Graphic3d_MaterialAspect(Graphic3d_NOM_PLASTIC)
Graphic3d_MaterialAspect.SetDiffuse(m, desired_diffusion_factor)

I already found that this SetDiffuse does not exist anymore in OCCT 7.4. Do you have any idea how to accomplish this in OCCT 7.4 and PythonOCC?

If this is not the appropriate place to ask this question feel free to say so!

Durksz, 2019-12-30:
An consequence of the setting the color as you advise is that the edges of the part also get this color. The previous version with OCE 6.9 the edges stayed black.

pridi, 2021-03-19:
@Durksz Did you find a way to change the color of the part but maintaining the black color of edges?

Durksz, 2021-03-19:
No, haven't found it.
I stopped searching and reluctantly accepted.