tpaviot/pythonocc-core #737: Error in importing module with 7.4.0beta2

zjiang0529, 2019-12-19:
Hello, 

Recently, I have built the source code of the new version pythonocc-core 7.4.0beta2. 
Before I have been using the pythonocc-core-0.18.1 version. 
And I notice that I can no longer call the module by using "OCC.BRep" for example, 
I should call modules by using "OCC.Core.BRep". This is not really an issue, however I 
got the following error when I import OCC.Core.BRep : No module named '_BRep'

Could you help me to fix this issue ? Thanks a lot. 

Zhifan Jiang

tpaviot, 2019-12-20:
What's your OS ? Did you also compiled opencascade by your own ? Can you please insert the full log ?

zjiang0529, 2019-12-20:
Thank you for your reply. I work on Ubuntu 18.04 LTS, with Python 3.7 and Conda environment. 
Yes, I also compiled opencascade-7.4.0 and I defined correctly the built occt-install directory in cmake 
when I was building pythonocc-core. What kind of log should I insert ?

zjiang0529, 2019-12-20:
I am wondering maybe it's a problem of libTKernel, I did not notice that before. 
[image]
[image]

tpaviot, 2019-12-20:
1. Be sure you add occt lib directory to the $PATH environment variable ;

2. Be sure the occt libraries in your $PATH are 7.4.0 (if ever you work with concurrent versions of occt, that might cause such issues)

tpaviot, 2019-12-20:
for instance, in my ```.bashrc```, I have inserted the following line
```
export LD_LIBRARY_PATH=/opt/build/occt740/lib:$LD_LIBRARY_PATH
```

zjiang0529, 2019-12-20:
Thanks a lot! I think that's why it could not find the lib files found in occt lib directory. 
I have added the occt lib directory in $PATH variable and the issue has been fixed. 
I have just one more question : instead of calling OCC.BRep, all the modules are packed in OCC.Core. Is this a new feature and a choice by your part for this new version ?

tpaviot, 2019-12-23:
It's a change in the package structure. You have to move all your ```import OCC.xxx``` to ```import OCC.Core.xxx```

closed 2019-12-23 by tpaviot