tpaviot/pythonocc-core #738: Possibility of JupyterRenderer webgl spacemouse support

Krande, 2019-12-20:
Hi,

Lately I have gotten to use the JupyterRenderer extensively in my daily work flow. Something I have been looking into is how to make use of my spacemouse navigator to rotate and move the 3d objects (like I am able to do in Abaqus, Freecad, Blender, etc..). 

I found a github repo (https://github.com/archilogic-com/aframe-space-navigator-controls) that might fit with the JupyterRenderer (assuming somehow the three.js implementation is transferable to pythreejs). At the same time as I attempt to make this work on my own, I thought I could check if anyone else have some experience with the this, or have some thoughts on how this might actually be implemented?

Kind Regards
Kristoffer

tpaviot, 2019-12-23:
Are you sure you posted to the correct issue tracker ? I guess it's rather something related to pythreejs https://github.com/jupyter-widgets/pythreejs

Krande, 2019-12-23:
Good point! I'll post the question there instead.

closed 2019-12-23 by Krande